xoseperez/espurna #2401: espurna 1.14.1 crashes when connecting to mqtt server

mciuraru, 2020-11-28:
**Bug description**
espurna 1.14.1 crashes when connecting to mqtt server

**Steps to reproduce**
Configure mqtt client in espurna and let it connect to mqtt server.

Output:
```
[005207] [MQTT] Subscribing to home/plugs/espurna-bruno/pulse/+/set (PID 3)
[005214] [MQTT] Subscribing to home/plugs/espurna-bruno/led/+/set (PID 4)
[005220] [MQTT] Subscribing to home/plugs/espurna-bruno/action/set (PID 5)
[005384] [MQTT] Subscribe ACK for PID 2
[005384] [MQTT] Subscribe ACK for PID 3
[005385] [MQTT] Subscribe ACK for PID 4
[005385] [MQTT] Subscribe ACK for PID 5

Exception (28):
epc1=0x4021e698 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys
sp: 3ffffc90 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe30:  00000000 00000000 3fff0420 3ffefdd0
3ffffe40:  4024c0d1 3ffffed0 4024bfb9 00000080
3ffffe50:  00000000 00000006 3fff76f4 402440ec
3ffffe60:  3fff6f34 7ef429e6 00000000 0000012c
3ffffe70:  3fff11f8 4050c000 00000000 4021e816
3ffffe80:  3fff6f34 00366fe6 3fff319c 3fff763c
3ffffe90:  00000000 3fff763c 00000000 4021e864
3ffffea0:  00000000 611a0000 3fff763c 4021c06a
3ffffeb0:  3fff7640 3fff76f4 00000000 4021c3ef
3ffffec0:  3fff11f8 00000000 00000020 fffffffc
3ffffed0:  00000000 0000141c 3fff11f8 4021c483
3ffffee0:  3fff3574 3fff3584 00000000 3fffdab0
3ffffef0:  00000000 00000000 3fff306c 40240d5c
3fffff00:  00000000 00000000 3fff763c 40240f01
3fffff10:  3fff28b8 000008da 000008da 00000000
3fffff20:  3fff2d78 3fff2d74 0019d81a 40240f50
3fffff30:  402326e8 3ffeef58 3fff3184 4024239c
3fffff40:  4024c6e2 3ffeef58 3ffeef58 4024253d
3fffff50:  40242394 00000000 00000000 00000031
3fffff60:  40237988 3ffef8d8 3ffef8f0 60000600
3fffff70:  00533147 3ffef8f0 3ffef8d8 40237995
3fffff80:  402379da 3fffdab0 00000000 3fffdcb0
3fffff90:  3ffef918 3fffdab0 00000000 4022f087
3fffffa0:  40000f49 40000f49 3fffdab0 40000f49
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000201] 

```
Tested on multiple SONOFF TH16 boards.
Exception (28) seem to indicate - Access to invalid address

This seem to be caused by async-mqtt-client#v0.8.1 
**When using the latest version of async-mqtt-client the device works as expected.**

**Device information**
```
[300564] [MAIN] ESPURNA 1.14.1
[300564] [MAIN] xose.perez@gmail.com
[300564] [MAIN] http://tinkerman.cat

[300564] [MAIN] CPU chip ID: 0x3B938B
[300566] [MAIN] CPU frequency: 80 MHz
[300570] [MAIN] SDK version: 1.5.3(aec24ac9)
[300573] [MAIN] Core version: 2.3.0
[300577] [MAIN] Core revision: 9826c6d
[300582] [MAIN] Build time: 1606585359
[300585]
[300585] [MAIN] Flash chip ID: 0x1440E0
[300588] [MAIN] Flash speed: 40000000 Hz
[300591] [MAIN] Flash mode: DOUT
[300594]
[300595] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[300603] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[300615] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[300617] [MAIN] Firmware size       :   496944 bytes /  122 sectors (   1 to  122)
[300624] [MAIN] Max OTA size        :   524288 bytes /  128 sectors ( 123 to  250)
[300631] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[300638] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[300645]
[300646] [MAIN] EEPROM sectors: 251, 250
[300650] [MAIN] EEPROM current: 250
[300657]
[300657] [MAIN] EEPROM:  4096 bytes initially |   354 bytes used ( 8%) |  3742 bytes free (91%)
[300664] [MAIN] Heap  : 33344 bytes initially | 20128 bytes used (60%) | 13216 bytes free (39%)
[300672] [MAIN] Stack :  4096 bytes initially |  1648 bytes used (40%) |  2448 bytes free (59%)
[300679]
[300680] [MAIN] Boot version: 31
[300683] [MAIN] Boot mode: 1
[300685] [MAIN] Last reset reason: Power on
[300691] [MAIN] Last reset info: flag: 0
[300694]
[300694] [MAIN] Board: ITEAD_SONOFF_TH
[300697] [MAIN] Support: API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL WEB
[300710] [MAIN] OTA: ARDUINO ASYNCTCP WEB
[300714] [MAIN] Sensors: DHTXX
[300718] [MAIN] WebUI image: SENSOR
[300720]
[300723] [MAIN] Firmware MD5: a0e2174e36dd8b86c5b0af9f5842c80f
[300728] [MAIN] Power: 3139 mV
[300729] [MAIN] Power saving delay value: 10 ms
[300733]

---8<-------

[300735] +OK

```
**Tools used**
* *Desktop operating system Windows 10 *
* *Firefox 83*
* *Platformio 2.2.1 VS 1.51.1*

mcspr, 2020-11-28:
Have you tried `dev` version instead of `1.14.1`?

There may be subtle differences in how PIO builds the firmware with master branch, see https://github.com/platformio/platformio-core/issues/3598
i.e. which version of the underlying ESPAsyncTCP is actually used with the resulting executable file. I wonder if master still crashes when lib_deps are reordered like in the current dev .ini, just make sure to clean-up $espurna\code\.pio\libdeps dir beforehand

Not sure I see any other issues based on https://github.com/marvinroger/async-mqtt-client/compare/v0.8.1...master

mciuraru, 2020-11-28:
Hello, I got he 1.14.1 release and compiled from there. The only modification done was arduino.h:

```
175c75
< //#define ITEAD_SONOFF_BASIC
---
> #define ITEAD_SONOFF_BASIC
163c163
< //#define ALEXA_SUPPORT          0
---
> #define ALEXA_SUPPORT          0
170,171c170,171
< //#define DEBUG_WEB_SUPPORT      0
< //#define DOMOTICZ_SUPPORT       0
---
> #define DEBUG_WEB_SUPPORT      0
> #define DOMOTICZ_SUPPORT       0
188c188
< //#define RPN_RULES_SUPPORT      0
---
> #define RPN_RULES_SUPPORT      1
194,195c194,195
< //#define THINGSPEAK_SUPPORT     0
< //#define TUYA_SUPPORT           0
---
> #define THINGSPEAK_SUPPORT     0
> #define TUYA_SUPPORT           0
211c211
< //#define DHT_SUPPORT            1
---
> #define DHT_SUPPORT            1
```

I've opened the issue just as a heads-up because I saw that in the dev branch, async-mqtt-client#v0.8.1 is still used, maybe it helps someone. As stated in the original text I've managed to get it working with the latest version of async-mqtt-client.
It might be something specific to the SONOFF TH16.... or maybe I've f@#$ something up. :)

mcspr, 2020-11-28:
You are in an unique position of reproducing it though, which I meant to point out. You may've also hit some weird bug that is already fixed in dev, which is only tangentially related to the mqtt client :) (i.e. the library issue, changed callbacks, anything else really). I have tried local server with master build similar to yours, but nothing bad happened there and it is still running stable. 

Exception trace is unique to the resulting .bin as well, so you'd need to decode it through something like https://github.com/me21/EspArduinoExceptionDecoder to translate addresses into some intelligible function names.

mciuraru, 2020-12-02:
Hello, what you are asking is way over what I am able to do :). I've build the latest dev version (which uses async-mqtt-client#v0.8.1 ) and works ok.
This might be limited to my setup/build env. and only on 1.141, so I guess the issues can be closed.

closed 2020-12-02 by mciuraru