xoseperez/espurna #2406: SONOFF RF - RF Module Missing ?

spiderjn, 2020-12-06:
i recently flash some Sonoff RF, but i can't see option to configure this feature.

i took : espurna-1.14.1-itead-sonoff-rf.bin

then when i ask INFO, i can't see option RF ?

[MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB

Where i can get a firmware including this build option ?

Best regards

mcspr, 2020-12-07:
RF variant simply 'presses' the button, everything related to radio signals is handled via an external chip
https://github.com/xoseperez/espurna/issues/1021#issuecomment-569934371

closed 2020-12-07 by mcspr

spiderjn, 2020-12-07:
ok, i tried with button, it's work but i want to configure my SONOFF RF without need a remote control (or a way to send a RF code from my HASS installation).

in WEB UI ESPURNA, on left menu i can't see "RF" entry ? 
(like in the previous sofware).

spiderjn, 2020-12-07:
i find how to generate a code from my HASS, 
you can close