xoseperez/espurna #2407: Sonoff POW non persistent after power outage

pcornill, 2020-12-07:
Espurna 1.14.2 (I tried 1.15.0 but then I have another issue: [#1038](https://github.com/xoseperez/espurna/issues/1038) 
I have setup the autosave to 5 reports. Saving is working and restore works after reboot BUT NOT after power outage (unplug / plug power) where it resets the Energy.

pcornill, 2020-12-15:
Anyone with the same issue? Is it a bug or designed that way?

mcspr, 2020-12-15:
Does it work after this change?
```diff
diff --git a/code/espurna/sensor.cpp b/code/espurna/sensor.cpp
index a8d73272..9261879b 100644
--- a/code/espurna/sensor.cpp
+++ b/code/espurna/sensor.cpp
@@ -2284,6 +2284,7 @@ void _sensorReport(unsigned char index, const sensor_magnitude_t& magnitude) {

 void _sensorInit() {

+    _sensor_save_every = getSetting("snsSave", SENSOR_SAVE_EVERY);
     _sensors_ready = true;

     for (auto& sensor : _sensors) {
```

pcornill, 2020-12-16:
Thanks for the suggestion but I'm on windows.
Is there a way to have a binary of the latest version?
The one available is 1.14.1 and is one year old.
Maybe I do not set the right parameters in config files?

Otherwise, if you can tell me what needs to be replaced in sensor.cpp?

Thanks for the great support.

mcspr, 2020-12-16:
> Otherwise, if you can tell me what needs to be replaced in sensor.cpp?

Just find the `_sensorInit()` func and insert:
```cpp
    _sensor_save_every = getSetting("snsSave", SENSOR_SAVE_EVERY);
```
at the start of it.

Result should look like the snippet above
(Without the `+`. It is there only to signify the 'addition' of the line to the file)

pcornill, 2020-12-17:
It was already there !

closed 2020-12-24 by mcspr

pcornill, 2020-12-24:
My issue is not fixed!

Closed #2407<https://github.com/xoseperez/espurna/issues/2407> via cd96f1c<https://github.com/xoseperez/espurna/commit/cd96f1ca6d8190d4db6d3a75d07cecb1083e92e7>.

mcspr, 2020-12-24:
Have you tried current build?

pcornill, 2020-12-24:
Crurent build I tried is 1.14.2. Is that the latest? It is one year old

mcspr, 2020-12-24:
Message above reads:
> With Espurna 1.15.0 Sonoff POW R1

So is it 1.14.2 or 1.15.0? 1.15.0 is the current development version, my assumption was that you built it yourself. I see the energy restored with a test device
You can wait for nightly (but it may not be available atm) or build yourself, just see the wiki for instructions

mcspr, 2020-12-24:
Just in time :) sry for delays

I also wanted to point out that Adafruit's neopixel lib might not be the most efficient one on the platform, you might want to check out either:
- https://github.com/FastLED/FastLED
- https://github.com/Makuna/NeoPixelBus

esp8266 code in the Adafruit was not really updated in any way since 2015

pcornill, 2020-12-24:
I tested both. The dev load with all default settings except the device type set to POW

mcspr, 2020-12-24:
But can you try again with the version that was just pushed?
https://github.com/xoseperez/espurna/commit/f166f6c7929108664d539f93a0da3e3fe0df36fa

pcornill, 2020-12-28:
You can close the issue, IT WORKS!
Thanks a lot

f166f6c<https://github.com/xoseperez/espurna/commit/f166f6c7929108664d539f93a0da3e3fe0df36fa>

commit cd96f1ca (2020-12-24): sns: load energy from settings unconditionally

fixes #2407
avoid depending on other settings

technically, it could be another block in the snsConfigure for
magnitudes, however it also should be a new type of one-shot logic