xoseperez/espurna #2409: sonoff 4ch pro R3 with RF

pcornill, 2020-12-15:
Hello, I saw the question #2372 but there is no mention if the RF433 is working.
I have one that I flashed with Espurna 1.15.0 but I do not see the RF menu.
I also tried Tasmota 9.1 but same issue, I cannot make it learn the RF codes.
Anyone succeed with RF?

mcspr, 2020-12-30:
As far as I understood, RF on Sonoff devices is still connected via an external hardware. Meaning, we don't receive RF codes directly, but receive a simple button presses and everything related to RF is handled by a separate MCU.

You may find the description in the manual about how to learn codes, but we can't really control that besides reacting to button presses as-if those are physical presses

pcornill, 2021-01-01:
It is working now. You can close the ticket. Thx

closed 2021-01-01 by mcspr