xoseperez/espurna #2413: Nodemcu 4channel firmware to use

bigjuanpa, 2021-01-01:
Hi, I need to use a nodemcu with a 4 channel relay.
I was not able to find what firmware so I need to use.
I teat a few and I was only able to use 1channel I can not add more switches.
Please what firmware I need to use for that board
Or how can I add that.
Thanks.

ColinShorts, 2021-01-01:
Pick one of the 4ch builds that have a normal relay per pin and use the pinouts of that device. ITEAD_SONOFF_4CH should work fine for testing purposes.

bigjuanpa, 2021-01-03:
Thanks for your reply, I was able to use the sonoff 4ch firmware but is inverted the power  when is on is off and with off is off.
Do you know how can I change that?

ColinShorts, 2021-01-03:
You'll need to build a binary yourself, look for examples of RELAY_TYPE_INVERSE in hardware.h - you'll need to set that for each relay.

bigjuanpa, 2021-01-03:
Thanks Colin for your reply.
I will try to do that I need to build a generic bin from the source code or I need to find the code for sonoff 4ch?
Thanks

ColinShorts, 2021-01-03:
Follow the guide on the wiki: "PlatformIO · xoseperez/espurna Wiki · GitHub" https://github.com/xoseperez/espurna/wiki/PlatformIO

You'll need to modify the source code as per above (use the 4ch, not the 4ch pro), from memory you will need to add 4 lines to set the inverse for each relay, but please take a little time to read through the file for examples.

#define RELAY1_TYPE         RELAY_TYPE_INVERSE
#define RELAY2_TYPE         RELAY_TYPE_INVERSE
#define RELAY3_TYPE         RELAY_TYPE_INVERSE
#define RELAY4_TYPE         RELAY_TYPE_INVERSE

closed 2021-01-14 by mcspr