xoseperez/espurna #2416: Seemed to have lost BUTTON support with ca15752

davebuk, 2021-01-16:
https://github.com/xoseperez/espurna/commit/ca15752f3b495322444f42f8f085357e78d0d2e6

I have just tried a custom sonoff basic and itead sonoff basic build and neither have support for buttons. The is nothing shown in the webUI. Am I missing something?

davebuk, 2021-01-17:
It's actually the relay that is missing, debug is showing the button presses, they just aren't switching the relay. MQTT is the same. There isn't an entry for 'SWITCHES' in the webUI.

mcspr, 2021-01-17:
What does `gpio` and `keys` commands output looks like?

davebuk, 2021-01-17:
Built using `ITEAD-SONOFF-BASIC` ENV.
`gpio`
```
* INPUT  @ GPIO00 (HIGH)
* INPUT  @ GPIO01 (LOW)
  INPUT  @ GPIO02 (HIGH)
* INPUT  @ GPIO03 (LOW)
  INPUT  @ GPIO04 (LOW)
  INPUT  @ GPIO05 (HIGH)
  INPUT  @ GPIO09 (LOW)
  INPUT  @ GPIO10 (LOW)
* INPUT  @ GPIO12 (LOW)
* OUTPUT @ GPIO13 (HIGH)
* INPUT  @ GPIO14 (HIGH)
  INPUT  @ GPIO15 (LOW)
  INPUT  @ GPIO16 (HIGH)
+OK
```
`keys`
```
Current settings:
> apiEnabled => "0"
> apiRealTime => "0"
> apiRestFul => "1"
> boardName => "ITEAD_SONOFF_BASIC"
> btnGPIO0 => "0"
> btnProv => "1"
> btnRelay0 => "0"
> btnRepDel => "500"
> cfg => "5"
> dns0 => "8.8.8.8"
> hbInterval => "300"
> hbMode => "2"
> ledGPIO0 => "13"
> ledInv0 => "1"
> ledMode0 => "2"
> ledRelay0 => "0"
> mqttEnabled => "1"
> mqttGroupSync0 => "0"
> mqttKeep => "300"
> mqttPort => "1883"
> mqttQoS => "1"
> mqttRetain => "1"
> mqttTopic => "SonoffB006"
> mqttUseJson => "0"
> mqttUseSSL => "0"
> ntpServer => "pool.ntp.org"
> ntpTZ => "CET-1CEST,M3.5.0,M10.5.0/3"
> relayBoot0 => "0"
> relayBootMask => "0b0"
> relayGPIO0 => "12"
> relayOnDisc0 => "0"
> relayPulse0 => "0"
> relaySync => "0"
> relayTime0 => "0"
> relayType0 => "0"
> sysScTraceMax => "128"
> telnetAuth => "1"
> telnetSTA => "0"
> wifiScan => "1"
> wsAuth => "1"
Number of keys: 53
```

mcspr, 2021-01-17:
Is there anything in the custom.h setting this GPIO?
```
* INPUT  @ GPIO12 (LOW)
```

It should say:
```
* OUTPUT @ GPIO12 (LOW)
```

Config is more strict atm, but I don't see any GPIO keys with value of "12"

mcspr, 2021-01-17:
Ah. Just `del relayType0`

commit 8e659c94 (2021-01-17): relay: type is always valid

resolve #2416
closed 2021-01-17 by mcspr