xoseperez/espurna #2418: POSIX TZ variable for NTP differs from Wiki.

davebuk, 2021-01-17:
I have been updating a lot of my devices and noticed the NTP section changed a while ago for the time zone setting. A lot of my devices currently show: `CET0CEST,M3.5.0,M10.5.0/3`. Looking at: https://github.com/xoseperez/espurna/wiki/NTP I set: `
#define NTP_TIMEZONE TZ_Europe_London` to one of my build settings. I thought this should change the time zone to: `GMT0BST,M3.5.0/1,M10.5.0`. I can put that time zone data in manually and espurna gets the correct time, but I wanted to specify it correctly in my `custom.h` builds

mcspr, 2021-01-18:
This must be part of the conversion happening here:
https://github.com/xoseperez/espurna/blob/8e659c94867bc41473ccbc45ff269f63fbd36721/code/espurna/ntp.cpp#L350-L399

If either of `ntpOffset`, `ntpDST` or `ntpRegion` are found in settings, it will construct a `ntpTZ` value based on those + old defaults for the NTP configuration

davebuk, 2021-01-18:
I'll try a loading a build to a factory reset device and check it loads the correct values.

davebuk, 2021-01-20:
https://github.com/xoseperez/espurna/pull/2414#issuecomment-763968914

The NTP server has always worked and been set to the default, but after loading a firmware and defining the time zone, the server address has been cleared. I can manually put it back in and it saves.

mcspr, 2021-01-21:
Current dev now includes these:
https://github.com/xoseperez/espurna/commit/c179467be27b4af990121ee22b6d95001cca61f0
https://github.com/xoseperez/espurna/commit/e8e58b821c88c9dac5bca41a40bb54b83a8b9923

Which should avoid:
- removing existing `ntpTZ` when old cfg vars are detected. I wonder though what is better - somewhat broken conversion (note the summer time transition hour is 2:00 by default, specified after the `/`) or just remove old settings and use UTC, forcing to convert manually through TZ.h
- trying to detect bad values from DHCP and disabling this feature via `ntpDhcp` setting
- reworked ntp startup to detect changes with both server and tz

davebuk, 2021-01-21:
After clearing out the `Time Zone` data from webUI and building my tuya dimmer device using `#define NTP_TIMEZONE			TZ_Europe_London` after boot, NTP keys still show: 
```
> ntpServer => "pool.ntp.org"
> ntpTZ => ""
```

mcspr, 2021-01-21:
That's a weird one. WebUI sets the key to an empty string instead of deleting.

mcspr, 2021-01-22:
...which was always the case for the web? I don't think I changed that between 1.14.0 and now (at least I hope so, can't find any `delSetting` in the web source code history). Technically, the expected 'erase-all' interaction is factory reset, which will erase the settings storage forcing each module to use build-time flags as defaults

davebuk, 2021-01-23:
A full factory reset has sorted it. It's cleared out all the keys and filled in the NTP settings I assumed it would correctly.

closed 2021-02-17 by davebuk