xoseperez/espurna #2419: FYI, random RELAY GPIO change when updating from older version.

davebuk, 2021-01-19:
I have some HBN wifi sockets that use the same config as the `LITESUN_LA_WF3` that have been running on 1.14.0. The RELAY is on `#12` BUTTON `#13` and two LEDs on `#4 and #5`.
1.14.0 `keys` gives:
```
[070768] > boardName => "HBN_BNC50"
[070771] > btnGPIO0 => "13"
[070776] > btnRelay0 => "0"
[070777] > btnRepDel => "0"
[070820] > ledGPIO0 => "4"
[070821] > ledInv0 => "1"
[070824] > ledMode0 => "7"
[070825] > ledMode1 => "8"
[070827] > ledRelay0 => "0"
[070830] > ledRelay1 => "0"
[070880] > relayBoot0 => "0"
[070885] > relayBootMask => "0b1111011"
[070886] > relayGPIO0 => "12"
[070890] > relayLastSch0 => "0"
[070891] > relayOnDisc0 => "0"
[070894] > relayPulse0 => "0"
[070897] > relaySync => "0"
[070899] > relayTime0 => "0"
[070902] > relayType0 => "0"
```

After building 1.15.0-dev8e659c94 with the same `custom.h` file, the relay wouldn't work and `keys` gave:
```
> boardName => "HBN_BNC50"
> btnGPIO0 => "13"
> btnRelay0 => "0"
> btnRepDel => "0"
> ledGPIO0 => "4"
> ledInv0 => "1"
> ledMode0 => "7"
> ledMode1 => "8"
> ledRelay0 => "0"
> ledRelay1 => "0"
> relayBoot0 => "0"
> relayBootMask => "0b1111011"
> relayGPIO0 => "5"
> relayLastSch0 => "0"
> relayOnDisc0 => "0"
> relayPulse0 => "0"
> relaySync => "0"
> relayTime0 => "0"
> relayType0 => "0"
```
For some reason the RELAY was set to the RED LED GPIO. The second LED option is missing from the webUI and there isn't a number next to the GPIO header in the SWITCHES section. it shows: `Switch #0 (gpio)`. A 1,14,0 device shows: `Switch #0 (GPIO12)`

Using `set relayGPIO0 12` and rebooting has restored the RELAY function and the second LED is now in the webUI and shown as configured as `#5`. I don't know why the RELAY GPIO should have changed. The other thing is that the second LED doesn't have any `key` entries for GPIO or INVERSE configuration.

mcspr, 2021-01-20:
Can you show `eeprom.dump`? At the end will be the raw representation of the settings

davebuk, 2021-01-20:
Minus WiFi stuff. Is there any other personal data in this code?
```
[348148] [WEBSOCKET] Requested action: dbgcmd

         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
------------------------------------------------------
0x0000:  FF 22 6D 61 63 22 3A 22 37 64 34 22 0A 88 70 72  "mac":"7d4"  pr
0x0010:  6F 64 5F 69 64 78 22 3A 22 30 32 37 35 31 32 34 od_idx":"0275124
0x0020:  31 22 2C 22 61 75 7A 5F 6B 65 79 22 3A 22 4B 79 1","auz_key":"Ky
0x0030:  51 74 6F 48 41 62 4B 61 50 34 44 49 6C 4E 78 6E QtoHAbKaP4DIlNxn
0x0040:  4E 6D 59 71 39 73 32 78 6C 34 6E 34 79 44 22 2C NmYq9s2xl4n4yD",
0x0050:  22 70 72 6F 64 5F 74 65 73 74 22 3A 66 61 6C 73 "prod_test":fals
0x0060:  65 7D 00 00 FF FF FF FF FF FF FF FF FF FF FF FF e}              
0x0070:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0080:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0090:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00A0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00B0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00C0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00D0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00E0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x00F0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0100:  D6 AF 03 00 02 09 C2 0F 23 48 00 00 00 00 00 00         #H      
0x0110:  00 00 C2 0F 23 48 00 00 00 00 80 ED FF 3F B0 FF     #H       ?  
0x0120:  FF 3F 80 00 1A 00 00 00 98 ED FE 3F 20 00 00 00  ?         ?    
0x0130:  D7 0D 10 40 72 10 10 40 A0 4E FF 3F 02 00 00 00    @r  @ N ?    
0x0140:  00 00 00 00 30 0B FF 3F F8 0A FF 3F 54 40 FF 3F     0  ?   ?T@ ?
0x0150:  1E 82 23 40 A4 5C 24 40 F8 0A FF 3F 54 40 FF 3F   #@ \$@   ?T@ ?
0x0160:  3C 86 23 40 F0 ED FF 3F 80 EE FF 3F 98 ED FE 3F < #@   ?   ?   ?
0x0170:  1A 00 00 00 0C E7 FE 3F 80 EE FF 3F 98 ED FE 3F        ?   ?   ?
0x0180:  54 40 FF 3F 54 40 FF 3F B0 44 FF 3F F8 0A FF 3F T@ ?T@ ? D ?   ?
0x0190:  11 00 24 40 01 00 5E 00 00 FB FF 3F 80 EE FF 3F   $@  ^    ?   ?
0x01A0:  9C 4E FF 3F FF FF FF FF FF FF FF FF FF FF FF FF  N ?            
0x01B0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x01C0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x01D0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x01E0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x01F0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0200:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0210:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0220:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0230:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0240:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0250:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0260:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0270:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0280:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0290:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02A0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02B0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02C0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02D0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02E0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x02F0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0300:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0310:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0320:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0330:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0340:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0350:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0360:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0370:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0380:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0390:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03A0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03B0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03C0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03D0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03E0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x03F0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
[... 167 lines cut ...]
0x0EA0:  68 62 49 6E 74 65 72 76 61 6C 00 0A 32 00 01 68 hbInterval  2  h
0x0EB0:  62 4D 6F 64 65 00 06 31 00 01 74 65 6C 6E 65 74 bMode  1  telnet
0x0EC0:  41 75 74 68 00 0A 30 00 01 74 65 6C 6E 65 74 53 Auth  0  telnetS
0x0ED0:  54 41 00 09 30 00 01 61 70 69 52 65 61 6C 54 69 TA  0  apiRealTi
0x0EE0:  6D 65 00 0B 31 00 01 61 70 69 52 65 73 74 46 75 me  1  apiRestFu
0x0EF0:  6C 00 0A 30 00 01 61 70 69 45 6E 61 62 6C 65 64 l  0  apiEnabled
[... cut ...]
0x0F90:  73 73 00 09 30 00 01 72 65 6C 61 79 54 79 70 65 ss  0  relayType
0x0FA0:  30 00 0A 30 00 01 62 74 6E 52 65 6C 61 79 30 00 0  0  btnRelay0 
0x0FB0:  09 31 33 00 02 62 74 6E 47 50 49 4F 30 00 08 34  13  btnGPIO0  4
0x0FC0:  00 01 6C 65 64 47 50 49 4F 30 00 08 30 62 31 31   ledGPIO0  0b11
0x0FD0:  31 31 30 31 31 00 09 72 65 6C 61 79 42 6F 6F 74 11011  relayBoot
0x0FE0:  4D 61 73 6B 00 0D 45 53 50 55 52 4E 41 2D 44 33 Mask  ESPURNA-D3
0x0FF0:  32 37 44 34 00 0E 68 6F 73 74 6E 61 6D 65 00 08 27D4  hostname  

+OK
```

mcspr, 2021-01-20:
Sry for not being explicit, my main interest were these lines at the 'end' of settings
```
0x0B70:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF                 
0x0B80:  FF FF FF FF FF FF FF FF FF FF FF FF FF 00 00 32                2
0x0B90:  35 35 00 03 73 00 00 FF FF FF FF 00 00 FF FF FF 55  s           
0x0BA0:  FF FF FF FF FF FF FF FF FF 00 00 FF FF FF FF FF                 
0x0BB0:  FF FF FF FF 00 00 47 4D 54 30 42 53 54 2C 4D 33       GMT0BST,M3
0x0BC0:  2E 35 2E 30 2F 31 2C 4D 31 30 2E 35 2E 30 00 18 .5.0/1,M10.5.0  
```
iirc, It was possible to accidentally read past the `00 00`  on, but on device those were already overwritten by the `ntpTZ` update and the manual `relayGPIO0` change. If you happen to have another ~1.14.0 device, you can try looking at where the GPIO keys were originally
(or, comment out https://github.com/xoseperez/espurna/blob/acf299582bc5d219607e9de262ea5120ba2f1856/code/espurna/ntp.cpp#L425, so no 'migration' code runs after the fact, eeprom.dump only works with telnet on the older version)

davebuk, 2021-01-30:
HBN device running 62ad7da3 `keys`
```
[171584] Current settings:
[171586] > adminPass => "***"
[171586] > board => "57"
[171586] > boardName => "HBN_BNC50"
[171587] > btnDelay => "0"
[171587] > btnGPIO0 => "13"
[171588] > btnRelay0 => "0"
[171591] > cfg => "3"
[171596] > desc => "Spare"
[171597] > dns0 => "8.8.8.8"
[171598] > gw0 => "192.168.1.1"
[171602] > hbInterval => "300"
[171603] > hbMode => "2"
[171605] > hostname => "ESPURNA-WiFi006"
[171609] > ip0 => "***"
[171612] > ledGPIO0 => "4"
[171617] > ledLogic0 => "1"
[171617] > ledMode0 => "7"
[171620] > ledMode1 => "8"
[171621] > ledRelay0 => "0"
[171624] > ledRelay1 => "0"
[171626] > mask0 => "255.255.255.0"
[171629] > mqttEnabled => "1"
[171632] > mqttGroupSync0 => "0"
[171637] > mqttPassword => "***"
[171638] > mqttQoS => "1"
[171640] > mqttServer => "***"
[171645] > mqttTopic => "WiFi006"
[171647] > mqttUser => "***"
[171650] > pass0 => "***"
[171653] > relayBoot0 => "0"
[171655] > relayBootMask => "0b1111011"
[171662] > relayGPIO0 => "5"
[171662] > relayLastSch0 => "0"
[171665] > relayOnDisc0 => "0"
[171667] > relayPulse0 => "0"
[171669] > relaySync => "0"
[171672] > relayTime0 => "0"
[171674] > relayType0 => "0"
[171677] > ssid0 => "***"
[171682] > wifiScan => "1"
[171683] > wsAuth => "1"
[171685] Number of keys: 41
[171689] Current EEPROM sector: 251
[171690] Free EEPROM: 3402 bytes (83%)
```
`eeprom.dump`
```
0x0C30:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
0x0C40:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 00
0x0C50:  32 35 35 00 03 73 63 68 53 77 69 74 63 68 00 00 255  schSwitch
0x0C60:  45 53 50 55 00 00 53 70 61 72 65 00 05 64 65 73 ESPU  Spare  des
0x0C70:  63 00 00 53 70 61 72 65 00 05 64 65 73 63 00 04 c  Spare  desc
0x0C80:  00 00 53 70 61 72 65 00 05 64 65 73 63 00 04 00   Spare  desc
0x0C90:  00 53 70 61 72 65 00 05 64 65 73 63 00 04 45 00  Spare  desc  E
0x0CA0:  00 53 70 61 72 65 00 05 64 65 73 63 00 00 00 53  Spare  desc   S
0x0CB0:  70 61 72 65 00 05 64 65 73 63 00 04 00 00 53 70 pare  desc    Sp
0x0CC0:  61 72 65 00 05 64 65 73 63 00 04 45 00 00 53 70 are  desc  E  Sp
0x0CD0:  61 72 65 00 05 64 65 73 63 00 04 45 53 50 55 52 are  desc  ESPUR
0x0CE0:  4E 41 2D 57 69 46 69 00 00 53 70 61 72 65 00 05 NA-WiFi  Spare
0x0CF0:  64 65 73 63 00 04 45 00 00 53 70 61 72 65 00 05 desc  E  Spare
0x0D00:  64 65 73 63 00 04 45 53 50 55 52 4E 41 00 00 53 desc  ESPURNA  S
0x0D10:  70 61 72 65 00 05 64 65 73 63 00 04 45 53 50 55 pare  desc  ESPU
0x0D20:  52 4E 41 2D 57 00 00 53 70 61 72 65 00 05 64 65 RNA-W  Spare  de
0x0D30:  73 63 00 04 45 53 50 55 52 4E 00 00 30 00 01 74 sc  ESPURN  0  t
0x0D40:  65 6C 6E 65 74 41 75 74 68 00 0A 31 00 01 74 65 elnetAuth  1  te
0x0D50:  6C 6E 65 74 53 54 41 00 09 31 39 32 2E 31 36 38 lnetSTA  192.168
0x0D60:  2E 31 2E 32 32 32 00 0D 6D 71 74 74 53 65 72 76 .1.***  mqttServ
0x0D70:  65 72 00 0A 48 42 4E 5F 42 4E 43 35 30 00 09 62 er  HBN_BNC50  b
0x0D80:  6F 61 72 64 4E 61 6D 65 00 09 57 69 46 69 30 30 oardName  WiFi00
0x0D90:  36 00 07 6D 71 74 74 54 6F 70 69 63 00 09 31 00 6  mqttTopic  1
0x0DA0:  01 6D 71 74 74 51 6F 53 00 07 37 37 37 52 21 32  mqttQoS  ***
0x0DB0:  79 6A 00 08 6D 71 74 74 50 61 73 73 77 6F 72 64 **  mqttPassword
0x0DC0:  00 0C 6F 70 65 6E 68 61 62 69 61 6E 00 0A 6D 71   ***  mq
0x0DD0:  74 74 55 73 65 72 00 08 31 00 01 6D 71 74 74 45 ttUser  1  mqttE
0x0DE0:  6E 61 62 6C 65 64 00 0B 30 00 01 62 74 6E 44 65 nabled  0  btnDe
0x0DF0:  6C 61 79 00 08 53 70 61 72 65 00 05 64 65 73 63 lay  Spare  desc
0x0E00:  00 04 45 53 50 55 52 4E 41 2D 57 69 46 69 30 30   ESPURNA-WiFi00
0x0E10:  36 00 0F 68 6F 73 74 6E 61 6D 65 00 08 30 00 01 6  hostname  0
0x0E20:  6C 65 64 52 65 6C 61 79 31 00 09 38 00 01 6C 65 ledRelay1  8  le
0x0E30:  64 4D 6F 64 65 31 00 08 37 00 01 6C 65 64 4D 6F dMode1  7  ledMo
0x0E40:  64 65 30 00 08 38 2E 38 2E 38 2E 38 00 07 64 6E de0  8.8.8.8  dn
0x0E50:  73 30 00 04 32 35 35 2E 32 35 35 2E 32 35 35 2E s0  255.255.255.
0x0E60:  30 00 0D 6D 61 73 6B 30 00 05 31 39 32 2E 31 36 0  mask0  192.16
0x0E70:  38 2E 31 2E 31 00 0B 67 77 30 00 03 31 39 32 2E 8.1.1  gw0  192.
0x0E80:  31 36 38 2E 31 2E 34 39 00 0C 69 70 30 00 03 64 168.1.**  ip0  
0x0E90:  62 77 69 72 65 6C 65 73 73 38 31 00 0C 70 61 73 ***  pas
0x0EA0:  73 30 00 05 42 72 6F 77 6E 4E 65 74 00 08 73 73 s0  ***  ss
0x0EB0:  69 64 30 00 05 31 00 01 77 69 66 69 53 63 61 6E id0  1  wifiScan
0x0EC0:  00 08 33 30 30 00 03 68 62 49 6E 74 65 72 76 61   300  hbInterva
0x0ED0:  6C 00 0A 32 00 01 68 62 4D 6F 64 65 00 06 31 00 l  2  hbMode  1
0x0EE0:  01 77 73 41 75 74 68 00 06 30 00 01 72 65 6C 61  wsAuth  0  rela
0x0EF0:  79 4F 6E 44 69 73 63 30 00 0C 30 00 01 6D 71 74 yOnDisc0  0  mqt
0x0F00:  74 47 72 6F 75 70 53 79 6E 63 30 00 0E 30 00 01 tGroupSync0  0
0x0F10:  72 65 6C 61 79 4C 61 73 74 53 63 68 30 00 0D 30 relayLastSch0  0
0x0F20:  00 01 72 65 6C 61 79 54 69 6D 65 30 00 0A 30 00   relayTime0  0
0x0F30:  01 72 65 6C 61 79 50 75 6C 73 65 30 00 0B 30 00  relayPulse0  0
0x0F40:  01 72 65 6C 61 79 42 6F 6F 74 30 00 0A 30 00 01  relayBoot0  0
0x0F50:  72 65 6C 61 79 53 79 6E 63 00 09 53 6F 6E 6F 66 relaySync  ***
0x0F60:  66 24 31 00 08 61 64 6D 69 6E 50 61 73 73 00 09 ***  adminPass
0x0F70:  30 00 01 72 65 6C 61 79 54 79 70 65 30 00 0A 35 0  relayType0  5
0x0F80:  00 01 72 65 6C 61 79 47 50 49 4F 30 00 0A 30 00   relayGPIO0  0
0x0F90:  01 62 74 6E 52 65 6C 61 79 30 00 09 31 33 00 02  btnRelay0  13
0x0FA0:  62 74 6E 47 50 49 4F 30 00 08 31 00 01 6C 65 64 btnGPIO0  1  led
0x0FB0:  4C 6F 67 69 63 30 00 09 34 00 01 6C 65 64 47 50 Logic0  4  ledGP
0x0FC0:  49 4F 30 00 08 35 37 00 02 62 6F 61 72 64 00 05 IO0  57  board
0x0FD0:  33 00 01 63 66 67 00 03 30 00 01 6C 65 64 52 65 3  cfg  0  ledRe
0x0FE0:  6C 61 79 30 00 09 30 62 31 31 31 31 30 31 31 00 lay0  0b1111011
0x0FF0:  09 72 65 6C 61 79 42 6F 6F 74 4D 61 73 6B 00 0D  relayBootMask

              [347739] +OK
```
But on switches webUI
`Switch #0 (GPIO12)` which is the GPIO defined in the `custom.h` file.

mcspr, 2021-01-30:
```
> board => "57"
```

Well, here it is from 1.14.1:
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/migrate.ino#L773-L782

iirc, this was done originally to support hw profiles in settings and those keys were written right after factory reset / initial setup based on the `cfg` key value and the build-time default

atm I simply delete those:
https://github.com/xoseperez/espurna/blob/05cc081b7ddb69bae76a41d3de1fc9d147bb6202/code/espurna/migrate.cpp#L90

(since https://github.com/xoseperez/espurna/commit/21b89efb4b60c98110dafe686c34ea49ea508e49)

davebuk, 2021-02-01:
Guess the issue is just down to a config miss match. You happy for me to close the issue?

mcspr, 2021-02-01:
Let me tweak migration once more, I will close it via PR / commit.

If you update the old device right now, relays will still use the broken config available in the moment, so device needs to be rebooted once again to re-read settings once again

commit 0f723b02 (2021-02-06): settings: handle old keys in the respective modules

fixes #2419
closed 2021-02-06 by mcspr