xoseperez/espurna #2420: Impossible to compile with BSEC Lib

itarika2, 2021-01-21:
[+]  searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
[+]  searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
[+] asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
[+] done the previous things again :)*

It's impossible to compile the project (lastest ESPurna dev version 660493a) 
with BSEC Lib (BME680 sensor) on Arduino IDE 1.8.13.

On ESP8266 **v2.6.0+** 

`/Users/User/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: espurna.ino.elf section '.text1' will not fit in region 'iram1_0_seg'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.`

On ESP8266 **v2.5.2-** 

`In file included from /Users/User/Documents/Arduino/espurna-dev/code/espurna/espurna.ino:48:0:
ntp.h:26:16: fatal error: TZ.h: No such file or directory
 #include <TZ.h>
                ^
compilation terminated.
exit status 1
TZ.h: No such file or directory`

### 🤷🏻‍♂️

mcspr, 2021-01-21:
Is it the version from https://github.com/BoschSensortec/BSEC-Arduino-library or from the library registry?
See #2402 and https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/BSEC-library-installation-instructions-for-ESP8266/td-p/14862

itarika2, 2021-01-21:
Oh, yes! BSEC library installation need additional instructions for ESP8266: *libalgobsec.a:(.literal.* .text.*)

Now it all works! ThankU!

itarika2, 2021-01-21:
Oh, yes! BSEC library installation need additional instructions for ESP8266: libalgobsec.a:(.literal. .text.*)

Now it all works! ThankU!

closed 2021-01-21 by itarika2

mcspr, 2021-01-21:
> Oh, yes! BSEC library installation need additional instructions for ESP8266: libalgobsec.a:(.literal. .text.*)

Not when using the most recent version though, see the Releases page on the library repo.