xoseperez/espurna #2421: Set default datetime on boot (use case without NTP)

itarika2, 2021-01-21:
In real life I use different firmware for esp8266 devices, all of them in local network only. So, I usually don't need NTP at all. But without time setting, devices goes to reboot every ±15 min.

My use case (for example in Tasmota) is simple rule: just set any default datetime on boot.

How to do this in espurna?

**Thanks!**

mcspr, 2021-01-21:
You mean with `NTP_SUPPORT` flag set to 0 it still tries to do... something with the time?

Scheduler is also ON implicitly, needs to be off as well since it only works via the clock time
https://github.com/xoseperez/espurna/blob/660493a1db47a4248f5cd34ebff584a9c1b126cc/code/espurna/config/general.h#L1602
https://github.com/xoseperez/espurna/blob/660493a1db47a4248f5cd34ebff584a9c1b126cc/code/espurna/config/general.h#L1566

itarika2, 2021-01-21:
@mcspr Yes, all these features are disabled, but the regular reboots still continues...

mcspr, 2021-01-22:
But how datetime is supposed to workaround that? 

We do still run DHCP <-> SNTP service under the hood. You can disable the SNTP service before it starts though,
inserting `sntp_servermode_dhcp(0);` somewhere in `setup()` func
https://github.com/xoseperez/espurna/blob/660493a1db47a4248f5cd34ebff584a9c1b126cc/code/espurna/main.cpp#L332

If reboot is caused by some kind of crash you should be able to see it via serial (when it happens) or via `crash` (or `crash force` if you already seen it) command after the fact, and decode the resulting trace via
https://github.com/me-no-dev/EspExceptionDecoder

itarika2, 2021-01-24:
@mcspr great, it works! Uptime - 48h without reboots and crashes.
Thank you!

closed 2021-01-24 by itarika2

mcspr, 2021-01-24:
Any chance you could decode the crash? I don't think it is something intended by the SDK :)