xoseperez/espurna #2422: tuya: physical button directly changes the lights state

mcspr, 2021-01-23:
After doing the factory reset on this device I had to re run: set tuyaChanState 1, set tuyaSwitch0 0 and set tuyaChannel0 2. LED had gone again even though tuya.show listed it and the button GPIO. tuya.save and a reboot has restored the LED in webUI without having to set the keys. Both LED and BUTTON GPIOs are also listed in the keys. The physical button turns the light ON/OFF but doesn't show it's state in the webUI or MQTT.

Debug data when pressing the button from an OFF state, the light then now turns ON at a dim level (webUI shows 102) and then pressing again which returns to OFF. The RED LED goes ON/OFF but the GREEN doesn't.
```
[912598] [TUYA] <=: 55aa0007000501010001010f
[912624] [TUYA] <=: 55aa00070008020200040000001026
[912636] [TUYA] <=: 55aa0007000501010001010f
[912648] [TUYA] <=: 55aa00070008020200040000003a50
[915380] [TUYA] =>: 55aa00000000ff
[915406] [TUYA] <=: 55aa000000010101
[916667] [TUYA] <=: 55aa0007000501010001000e
[916681] [TUYA] <=: 55aa00070008020200040000003a50
[916694] [TUYA] <=: 55aa0007000501010001000e
[916707] [TUYA] <=: 55aa00070008020200040000001026
```

_Originally posted by @davebuk in https://github.com/xoseperez/espurna/issues/2414#issuecomment-765887898_

mcspr, 2021-01-23:
It is expected, as we ignore the received state messages via tuyaFilter => 1
But...
- we always need to ignore cold boot state
- received channel value is only true half the time, first it's 16 (0x10) then 58 (0x3a)
- received relay state is off, while channel value is still on

Repeating my previous q, does removing relay boot state preservation (our setting, via Switches tab) and pressing the button to turn the light off preserve it on the next boot?

davebuk, 2021-01-23:
Relay boot is set to `Always OFF`. `tuyaFilter` doesn't appear in the keys so I have now added it. Current keys:
```
[252444] [WEBSOCKET] Requested action: dbgcmd
Current settings:
> adminPass => "***"
> boardName => "TUYA_GENERIC_DIMMER"
> brightness => "255"
> btnGPIO0 => "0"
> cfg => "5"
> ch0 => "102"
> dns0 => "192.168.1.1"
> gw0 => "192.168.1.1"
> hostname => "ESPURNA-A6CCBD"
> ip0 => "192.**""
> ledGPIO0 => "14"
> ledMode0 => "2"
> ltRelay => "1"
> ltStep => "100"
> mask0 => "255.255.255.0"
> mireds => "0"
> mqttClientID => "Tuya-001"
> mqttEnabled => "1"
> mqttPassword => "***"
> mqttQoS => "0"
> mqttServer => "192.***"
> mqttTopic => "TUYA001"
> mqttUser => "***"
> pass0 => "***"
> ssid0 => "**""
> tuyaChanState => "1"
> tuyaChannel0 => "2"
> tuyaFilter => "1"
> tuyaSwitch0 => "0"
> useColor => "0"
> useTransitions => "1"
Number of keys: 31
Available: 3291 bytes (85%)
+OK
```
After a cold boot, the light goes through its ON/OFF cycle. Once fully booted, pressing the button cycles the light ON/OFF but the webUI doesn't show the same status as the actual output.

davebuk, 2021-02-24:
I've looked at the hardware again and found that the physical button connects to ground via resistor R16 directly to the MCU PIN 10.

There is some info about this at: https://tasmota.github.io/docs/TuyaMCU-Devices/ `DM_WF_MDV4 Leading edge dimmer`.

Can espurna use the signal to the MCU PIN to operate like a standard GPIO type input?

I may look to modify the hardware, remove R16 and connect the push button directly to one of the TYWE3S unused GPIOs.
[image]

mcspr, 2021-02-24:
Maybe? If it is interpreted by the MCU as ON, that's the explanation as to why it wants the lights ON on boot. As-is we can't distinguish between the button press and echo MCU sends when it acknowledges the state we send to it (relay toggle, brightness, everything sends it back). TUYA only allows this condensed API, there's no direct GPIO reads / writes and MCU is expected to handle the inputs and outputs

btw, if you remember the previous issue:
https://github.com/xoseperez/espurna/issues/1729#issuecomment-498419304
stock fw does not seem to do anything different with communication

davebuk, 2021-02-24:
The button is labelled on the outside as 'RESET'. Maybe that's all the original firmware did?

mcspr, 2021-02-24:
:shrug:
It is just toggling and sending the state though, based on the logs above. There's no reset notification incoming

davebuk, 2021-02-24:
I meant the original TUYA firmware.

mcspr, 2021-02-24:
I don't get what you mean though, the original fw receives the same status message on the press so .

From ESP side we receive only two GPIOs to work with:
- GPIO14 as a status LED, which we write to the LED action stuff
- GPIO0 button, which is not really connected to anything? as the real button is managed by the MCU itself. Based on the docs, it is a signal for the ESP to reset the WiFi

As a wild guess - does changing LED status to the relay follow mode changes anything on boot? Meaning, it kept low at boot. I don't remember if this was tested

davebuk, 2021-02-24:
Sorry, I didn't know if you knew what the original firmware was doing with the button press. From what you are saying, it WOULD have toggled the output, the same as espurna does.

I'll look at the LED later.

mcspr, 2021-02-24:
Flashing original .bin will also clear out the remaining questions about the button? Just note that going back to espurna would need a full flash clear to be safe from SDK config remains

davebuk, 2021-02-24:
Using firmware built around end of Jan 2021, the LED was set to `follow switch`. After a cold boot, the GREEN LED flashed ON then OFF.

On a side note, the device had been removed from the mains for a few days. On power up, the light went back to the old issue where it would turn itself ON and stay ON with the RED LED ON. Looking at a review on amazon (https://www.amazon.co.uk/gp/customer-reviews/R2ZERWRHD6MTQL/ref=cm_cr_dp_d_rvw_ttl?ie=UTF8&ASIN=B07Q2XSYHS) this seems to be a standard issue with the tuya firmware. It did this every time I cold booted. I noticed in the webUI the channel value was at `0` even though it was last left at `255`. Setting this to `255` and carrying out a cold boot has restored the "auto OFF" function on cold boot.

Due to recent changes in espurna, I built using 7434ea3. I have lost LED from the webUI so tried the same procedure as per the top of this thread. I still didn't have LED but cold boot "auto OFF" now works regardless of channel value before cold boot. 

I decided to do a factory reset from the webUI to clear out all settings and start again. In doing so I now can't get to the web interface! I can connect to the espurna AP using the default password but the webUI on 192.168.4.1 doesn't open. I may have to look at reflashing using the hardware pins.

davebuk, 2021-02-25:
I flashed the .bin shown here: https://github.com/xoseperez/espurna/issues/1729#issuecomment-498419304. I didn't setup the device with the Smart Life software but confirmed the button does toggle the light output. The GREEN LED flashed ON then OFF on boot and then was always ON. That might be controllable once the device was setup.

I then uploaded `7434ea3` but still couldn't get into the webUI and setup the initial settings. I could connect to the ESPURNA AP but the webUI wouldn't load. I could connect using TELENT but didn't try anything else.

I have uploaded a previous version `660493a` and can access the webUI again. I'll look to upload `7434ea3` OTA later.

mcspr, 2021-02-25:
Thanks! The issue with initial boot is a big of lights code that thinks the lights did not change, because all values stay at 0 so it thinks there is nothing to do (but, I thought I already did in that latest commit...)

WebUI / WiFi issues without mains are kind of expected, I'd only try when connected to AC b/c of the power requirements. If it is already mains, I'd try to wait ~20min without power and then try to connect it to the AP.

davebuk, 2021-02-25:
All testing was done with mains not via USB device.

davebuk, 2021-02-25:
OTA'd back to 7434ea3. I realise why I may have lost LED. I see you now define the LED GPIO in the build definition rather than it being, I think before, discovered? I still had to `set ledGPIO0 14` to get the webUI to show, but don't think it's too much of an issue. Probably where I played around.

However, after a reboot, MQTT brightness changes were all over the place. Checked webUI and the `brightness` was at 83711 fully right.
`[031014] [MQTT] Sending TUYA001/brightness => 83711 (PID 1)`. Clicking on a random part of the slider, it returned to a normal level and then when slid fully right, showed 255. It happens every time I reboot. Tried setting `brightness` to 100 rather than 255. After reboot value was 83556.

mcspr, 2021-02-25:
```diff
diff --git a/code/espurna/light.cpp b/code/espurna/light.cpp
index e2ec0a06..08e5c2be 100644
--- a/code/espurna/light.cpp
+++ b/code/espurna/light.cpp
@@ -1041,7 +1041,7 @@ struct LightRtcmem {
     // - aabbccddee are channels (from 0 to 5 respectively)
     explicit LightRtcmem(uint64_t value) {
         _mireds = (value >> (8ull * 6ull)) & 0xffffull;
-        _brightness = (value >> (8ull * 5ull));
+        _brightness = (value >> (8ull * 5ull)) & 0xffull;

         _channels[4] = static_cast<uint8_t>((value >> (8ull * 4ull)));
         _channels[3] = static_cast<uint8_t>((value >> (8ull * 3ull)));
@@ -1065,8 +1065,8 @@ struct LightRtcmem {
     {}

     uint64_t serialize() const {
-        return (static_cast<uint64_t>(_mireds) << (8ull * 6ull))
-            | (static_cast<uint64_t>(_brightness) << (8ull * 5ull))
+        return ((static_cast<uint64_t>(_mireds) & 0xffffull) << (8ull * 6ull))
+            | ((static_cast<uint64_t>(_brightness) & 0xffull) << (8ull * 5ull))
             | (static_cast<uint64_t>(_channels[4]) << (8ull * 4ull))
             | (static_cast<uint64_t>(_channels[3]) << (8ull * 3ull))
             | (static_cast<uint64_t>(_channels[2]) << (8ull * 2ull))
```

Yeah, about that... there was a missing numeric operation, patch above should handle that

mcspr, 2021-02-25:
LED though. What happened before...
- before the January patch, when discovery happened it saved the `ledGPIO0` with the received value of 14
- after, it simply cached that value for `tuya.save` & `tuya.show` commands
- also, which you already noticed, generic dimmer config now simply has the LED1 pin

davebuk, 2021-02-25:
Above code change works.

So LED doesn't show up in tuya stuff now, just standard configs?

mcspr, 2021-02-25:
Tuya does not run discovery if there are any configuration flags present with DP != 0. I did shuffle things a lot that time, but this should still be possible:
```diff
diff --git a/code/espurna/tuya.cpp b/code/espurna/tuya.cpp
index 5389ad36..8f280b1e 100644
--- a/code/espurna/tuya.cpp
+++ b/code/espurna/tuya.cpp
@@ -368,13 +368,9 @@ error:
                     setupSwitches();
                 }

-                if (!configDone) {
-                    DEBUG_MSG_P(PSTR("[TUYA] Starting discovery\n"));
-                    state = State::QUERY_PRODUCT;
-                    return;
-                }
-
-                state = State::IDLE;
+                DEBUG_MSG_P(PSTR("[TUYA] Starting discovery\n"));
+                state = State::QUERY_PRODUCT;
+                return;
             }

             sendWiFiStatus();
```

On the first received heartbeat, it will read from settings and run normal discovery, which will read both GPIO values into the cache (at least)

davebuk, 2021-02-25:
Ah, ok. I thought it was the TUYA code that was doing the discovery and setting the dummy relay output for the dimmer as well as button and LED etc. All seems to be working at the moment. 😁

mcspr, 2021-02-26:
At least we got one over the stock :) It does not stay ON when power-cycled (unless requested)
I really don't see an easy way to implement the button though, so this issue is still on hold

davebuk, 2021-02-26:
And the fact espurna makes the device more secure and configurable.

davebuk, 2021-02-26:
If I were to re-configure the hardware push button as default high going low on press, directly to one of the spare ESP pins, do you imagine it would effect the TUYA side of things? 

Pin 10 of the MCU may need to be pulled high and say `button2` defined at build. Could I then assign that `button` to the light relay and use the actions `double-click`, `long-click` etc to control local `DIM UP`, `DIM DOWN` etc?

mcspr, 2021-02-26:
Or cut the trace, and connect the button directly to one of the available pins, replacing the GPIO0 button? (that does not do anything anyway, really)

davebuk, 2021-03-01:
It doesn't seem to be working. I've cut the track so the switched output of the button no longer connects to the MCU via R16. I built adding to my custom.h the definitions below but nothing is shown on the debug screen when I press the button. GPIOs 12 and 13 do not seem to be pulled up and have only 0.2V on the pin, going to 0V when I press the button. I tried with an external pullup resistor onto the GPIO and still nothing happened. Do I need to define a dummy relay?
```
// Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON2_PIN         12       //USE GPIO12 for push button
    #define BUTTON2_CONFIG      BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_RELAY       1
    #define BUTTON2_PRESS       BUTTON_ACTION_NONE
    #define BUTTON2_CLICK       BUTTON_ACTION_NONE
    #define BUTTON2_DBLCLICK    BUTTON_ACTION_DIM_DOWN
    #define BUTTON2_TPLCLICK    BUTTON_ACTION_DIM_UP
    #define BUTTON2_LNGCLICK    BUTTON_ACTION_TOGGLE
    #define BUTTON2_LNGLNGCLICK BUTTON_ACTION_TOGGLE
    #define BUTTON3_PIN         13       //USE GPIO13 for push button
    #define BUTTON3_CONFIG      BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    //#define BUTTON3_RELAY       1
    #define BUTTON3_PRESS       BUTTON_ACTION_NONE
    #define BUTTON3_CLICK       BUTTON_ACTION_NONE
    #define BUTTON3_DBLCLICK    BUTTON_ACTION_DIM_DOWN
    #define BUTTON3_LNGCLICK    BUTTON_ACTION_TOGGLE
    #define BUTTON3_LNGLNGCLICK BUTTON_ACTION_TOGGLE
```

I even removed the GREEN LED leg and removed the LED definition and re-configured `BUTTON2` to GPIO 14 in case 12 and 13 aren't actually wired to the board of the TYWE3S. I'm not sure why the GPIOs configured as a pullup push button don't have 3VDC on their pins?

I also added some to the `keys` for `BUTTON2` (GPIO/TYPE/MODE/PINMODE/PRESS) to the debug menu and still the GPIO is low and doesn't do anything. Any ideas?

mcspr, 2021-03-03:
Do basics work? Given these 3 settings on a basic dev board without any other config:
```
get btnGPIO0 btnPinMode0 btnDefVal0
> btnGPIO0 => "14"
> btnPinMode0 => "1" (aka InputPullup)
> btnDefVal0 => "1" (aka HIGH)
```
```
gpio
...
* INPUT  @ GPIO14 (HIGH)
...
+OK
```
Touching GND with the pin (D5 in case of a wemos board)
```
[064494] [BUTTON] Button #0 event 1 (pressed)
[064900] [BUTTON] Button #0 event 2 (released)
```

Note that the board *may* have external pull up or downs. There is also GPIO0 exposed that is definitely already pulled up by default

davebuk, 2021-03-03:
As in basic functions of the ESP on the TUYA board? If so.....
Button#1 not defined any where.
```
get btnGPIO0 btnPinMode0 btnDefVal0
> btnGPIO0 =>
> btnPinMode0 =>
> btnDefVal0 =>
```
`Button 2 and 3` defined in custom.h and the keys below for `Button 2` were placed by me in the debug screen.
```
get btnGPIO2 btnPinMode2 btnDefVal2
> btnGPIO2 => "14"
> btnPinMode2 => "1"
> btnDefVal2 => "1"
```
```
get btnGPIO3 btnPinMode3 btnDefVal3
> btnGPIO3 =>
> btnPinMode3 =>
> btnDefVal3 =>
```

KEYS
```
Current settings:
> adminPass => "***"
> boardName => "TUYA_GENERIC_DIMMER"
> brightness => "255"
> btnDefVal2 => "1"
> btnGPIO2 => "14"
> btnGPIOType2 => "1"
> btnMode2 => "0"
> btnPinMode2 => "1"
> btnPress2 => "1"
> cfg => "5"
> ch0 => "89"
> dns0 => "192.168.1.1"
> gw0 => "192.168.1.1"
> hostname => "ESPURNA-***"
> ip0 => "***"
> ledMode0 => "2"
> ltStep => "100"
> mask0 => "255.255.255.0"
> mireds => "326"
> mqttClientID => "Tuya-001"
> mqttEnabled => "1"
> mqttPassword => "***"
> mqttServer => "***"
> mqttTopic => "TUYA001"
> mqttUser => "***"
> pass0 => "***"
> ssid0 => "***"
> tuyaChanState => "1"
> tuyaChannel0 => "2"
> tuyaSwitch0 => "0"
> useColor => "0"
> useTransitions => "1"
Number of keys: 32
Available: 3267 bytes (85%)
+OK
```
```
gpio
  INPUT  @ GPIO00 (HIGH)
  INPUT  @ GPIO01 (LOW)
  INPUT  @ GPIO02 (HIGH)
  INPUT  @ GPIO03 (LOW)
  INPUT  @ GPIO04 (HIGH)
  INPUT  @ GPIO05 (HIGH)
  INPUT  @ GPIO12 (HIGH)
  INPUT  @ GPIO13 (LOW)
  INPUT  @ GPIO14 (HIGH)
  INPUT  @ GPIO15 (LOW)
  INPUT  @ GPIO16 (LOW)
+OK
```
With hardware button held down connected to GPIO14
```
gpio
  INPUT  @ GPIO00 (HIGH)
  INPUT  @ GPIO01 (LOW)
  INPUT  @ GPIO02 (HIGH)
  INPUT  @ GPIO03 (LOW)
  INPUT  @ GPIO04 (HIGH)
  INPUT  @ GPIO05 (HIGH)
  INPUT  @ GPIO12 (HIGH)
  INPUT  @ GPIO13 (LOW)
  INPUT  @ GPIO14 (LOW)
  INPUT  @ GPIO15 (LOW)
  INPUT  @ GPIO16 (LOW)
+OK
```
No button events are shown in the debug screen.

Interestingly after button release, GPIO14 is still LOW.

Next I `set btnGPIO2 0`. Although `gpio` is reporting the HIGH/LOW status correctly, there still isn't any button events in debug.

mcspr, 2021-03-03:
That is why my advice is to remove all existing config and simply try to modify the 1st button. `set btnGPIO2 0`, but you still have button on GPIO0; see `button` command.

mcspr, 2021-03-04:
OT: And about `get`... the original settings model is heavily dependent on the webui, since the code itself tries to use build settings first and only then all the serialized strings from settings. I think I'll add something like this:
```
get relayGPIO0 relayType0 relayProv0
> relayGPIO0 => 12 (default)
> relayType0 => normal (default)
> relayProv0 => gpio (default)
+OK
```
So build flag values can be displayed instead of the current blank spaces. There's already a code in place, just some minor tweaks are needed to actually make this work (...some runtime increase is expected as well...)

davebuk, 2021-03-04:
Rebuilt without any buttons defined and also built a 'generic tuya dimmer' both at f013e04. `del`'d all the keys from my custom.h build and added `set btnGPIO2 0` to both. On the 'generic tuya dimmer' and my 'custom.h' dimmer, both show the HIGH/LOW `GPIO0` state change but nothing in the debug screen for `BUTTONS`
.

mcspr, 2021-03-04:
`buttons` should show active buttons in the mean time. And I still don't get why there are 3 buttons now :shrug:
Are you using button contacts or trying to touch GND wire with the pin? And neither work?

davebuk, 2021-03-04:
Sorry, I mis-understood what you meant about `button` command. I thought I should be looking for the debug output like:
```
[064494] [BUTTON] Button #0 event 1 (pressed)
[064900] [BUTTON] Button #0 event 2 (released)
```
Anyway, using the default `tuya-dimmer` build, after typing `button` in debug and pulsing a wire from GND onto GPIO0, I **did** get outputs like above showing the press/released events as well as the MQTT messages. I tried to change the GPIO to 14 and although it did give the same output when I pressed the physical button soldered to GPIO 14, I kept getting re-starts. I think GPIO 14 is being externally pulled LOW, maybe through the other side of the LED? I'll look to change it back to GPIO 12 or 0.

I then tried to reload my custom.h version setting for button 2 on GPIO 14. I did have 2 and 3 defined so I could see which, if any, GPIOs are working on the board. Now typing `button` just gives an error (-ERROR: Command not found). Same when reloading the 'generic tuya dimmer'. Its the same after putting in `set btnGPIO2 0`.

I'll look to re-configure the hardware, set the LED back to 14 and try the button on 0. I know that is HIGH when OFF.

davebuk, 2021-03-04:
LED reconnected back to `GPIO 14` and button connected to `GPIO 0`. HIGH/LOW state is being shown when running `gpio` but `button` still gives an error and there are no button events.

`custom.h` button additions:

```
    // Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON2_PIN         0       //USE GPIO0 for push button
    #define BUTTON2_CONFIG      BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    #define BUTTON2_RELAY       1
    #define BUTTON2_PRESS       BUTTON_ACTION_NONE
    #define BUTTON2_CLICK       BUTTON_ACTION_NONE
    #define BUTTON2_DBLCLICK    BUTTON_ACTION_DIM_DOWN
	//#define BUTTON2_TPLCLICK    BUTTON_ACTION_DIM_UP
    #define BUTTON2_LNGCLICK    BUTTON_ACTION_TOGGLE
    #define BUTTON2_LNGLNGCLICK BUTTON_ACTION_TOGGLE
```
(TPLCLICK commented out as i'm not sure if that is an option?)

`help` gives:
```
Available commands:
> ADC
> BRIGHTNESS
> CHANNEL
> COMMANDS
> CONFIG
> CRASH
> DEBUG.BUFFER
> DEL
> EEPROM.COMMIT
> EEPROM.DUMP
> EEPROM
> ERASE.CONFIG
> FACTORY.RESET
> FLASH.DUMP
> GET
> GPIO
> HEAP
> HELP
> HOST
> HSV
> INFO
> KELVIN
> KEYS
> LIGHT
> MIRED
> MQTT.INFO
> MQTT.RESET
> NETSTAT
> NTP.SETTIME
> NTP
> OTA
> RELAY
> RELOAD
> RESET
> RGB
> RTCMEM.DUMP
> RTCMEM.REINIT
> SET
> STACK
> TUYA.SAVE
> TUYA.SHOW
> UPTIME
> WIFI.STATIONS
> WIFI
> WIFI.AP
> WIFI.RESET
> WIFI.SCAN
> WIFI.STA
+OK
```
I guess `BUTTON` should appear in that list?

davebuk, 2021-03-04:
Just run `tuya.show` 
```
Config:
"btnGPIO0" => "0"
"ledGPIO0" => "14"

Known DP(s):
1 (bool) => lights state
2 (int) => 0 (channel)
```
and `tuya.save` REBOOT.

Now have:
```
0 - GpioPin @ GPIO00
+OK
[017321] [BUTTON] Button #0 event 1 (pressed)
[017323] [MQTT] Sending TUYA001/button/0 => pressed (PID 1)
[017831] [BUTTON] Button #0 event 3 (click)
[017833] [MQTT] Sending TUYA001/button/0 => click (PID 1)
[019864] [TUYA] =>: 55aa00000000ff
[019889] [TUYA] <=: 55aa000000010101
[021147] [BUTTON] Button #0 event 1 (pressed)
[021149] [MQTT] Sending TUYA001/button/0 => pressed (PID 1)
[021656] [BUTTON] Button #0 event 3 (click)
[021658] [MQTT] Sending TUYA001/button/0 => click (PID 1)
[022765] [BUTTON] Button #0 event 1 (pressed)
[022767] [MQTT] Sending TUYA001/button/0 => pressed (PID 1)
[023274] [BUTTON] Button #0 event 3 (click)
[023276] [MQTT] Sending TUYA001/button/0 => click (PID 1)
```

Actions don't work as defined in my `custom.h` as long-click doesn't toggle and longlong-click reboots.

mcspr, 2021-03-07:
re. https://github.com/xoseperez/espurna/issues/2422#issuecomment-790961522
`button` command does not show up when there are 0 buttons, and one way it may happen is when GPIO is already used by something else / this routine fails:
https://github.com/xoseperez/espurna/blob/f013e04e2206be7b2cd7fa86e65021630337808a/code/espurna/button.cpp#L891-L893

There is no 1st button, unless you have BUTTON1_PIN or btnGPIO0, so subsequent button config is simply ignored. Which is why I suggested to simply change btnGPIO0 and reboot to apply the new pin. Plus, maybe change btnPinMode0 to 2 to make default state what it sees on boot, since it probably treats it as a click of an already pressed button when you connect the wire:
https://github.com/xoseperez/espurna/blob/f013e04e2206be7b2cd7fa86e65021630337808a/code/espurna/config/defaults.h#L212
https://github.com/xoseperez/espurna/blob/f013e04e2206be7b2cd7fa86e65021630337808a/code/espurna/config/defaults.h#L237

davebuk, 2021-03-08:
Its working! I hadn't set `BUTTON1` as I thought that was already being used by the tuya side of things so started with `BUTTON2, BUTTON3` etc. for testing.

I realised that triple click wasn't right
`define BUTTON1_TPLCLICK    BUTTON_ACTION_DIM_UP`
and worked out the following was correct.
`define BUTTON1 TRIPLECLICK    BUTTON_ACTION_DIM_UP`

Final button definitions:
```
    // Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON1_PIN  0     //USE GPIO0 for push button
    #define BUTTON1_CONFIG  BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_BOOT
    #define BUTTON1_RELAY 1
    #define BUTTON1_PRESS BUTTON_ACTION_NONE
    #define BUTTON1_CLICK  BUTTON_ACTION_NONE
    #define BUTTON1_DBLCLICK  BUTTON_ACTION_DIM_DOWN
    #define BUTTON1_TRIPLECLICK  BUTTON_ACTION_DIM_UP
    #define BUTTON1_LNGCLICK  BUTTON_ACTION_TOGGLE
    #define BUTTON1_LNGLNGCLICK  BUTTON_ACTION_TOGGLE
```

The final issue is that `DIM_UP/DIM_DOWN` adjusts the `BRIGHTNESS` not `CHANNEL#0`

davebuk, 2021-03-09:
My `CHANNEL#0` vs `BRIGHTNESS` issue can probably be solved by changing how I control it via MQTT. I'm currently adjusting the `CHANNEL#0` value rather than the  `BRIGHTNESS`. I guess `BRIGHTNESS` is the best value to change?

mcspr, 2021-03-09:
Either works. Changing brightness will also change the reported channel value though, since it is scaled

davebuk, 2021-03-09:
But using `CHANNEL` for MQTT doesn't change the value of 'brightness' but `DIM_UP/DOWN` does. So MQTT has to change `BRIGHTNESS` otherwise they get out of sync. If brightness gets set to 50%, setting channel to 100% is still 50% of the actual output.

davebuk, 2021-03-11:
Do you want to close this issue? The fix requires a hardware modification and addition of button definitions. Is there any harm in defining the `BUTTON` and `DIM_UP/DOWN/TOGGLE` actions in the `hardware.h` file so it is ready if people want to cut the PCB track and attach a wire to GPIO0?

mcspr, 2021-03-13:
Perhaps with a comment? I'd close this tbh, discussion went a bit too long a there's much OT to parse out :)

I wonder if other implementations of tuya platform do this initial send that causes the issue though:
https://templates.blakadder.com/aofo_4AC+4USB_tuya.html
But my hunch is, here it is caused because devs tried to sync up the brightness value... And it is best to use the hw mod instead, at least for this device

closed 2021-03-13 by mcspr

davebuk, 2021-03-15:
I'll look to add the device to the wiki at some point and add a few notes and pictures.

davebuk, 2021-06-16:
**WIP**
Modification to the board to allow ON/OFF/DIMMING using physical button. 
[image]

davebuk, 2021-06-16:
On the rear of the board, cut the track between the output of the switch and R16.
[image]

On the front of the board, solder a jumper wire between the switch leg and `GPIO0` on the TYWE3S.

davebuk, 2021-06-16:
On the rear of the board, cut the track between the output of the switch and R16.

[image]

On the front of the board, solder a jumper wire between the switch leg and `GPIO0` on the TYWE3S.

[image]

I used the following build definitions for control. It allows toggling the light ON/OFF with either a `LONG` or `LONG_LONG` button press, DIM DOWN with `DOUBLE` button click and BRIGHTEN UP with `TRIPLE` button click. I disabled single `PRESS` and `CLICK` toggles so I didn't have any debouce issues when trying to do double and triple clicks. These could end up toggling the light ON/OFF when I actually wanted to adjust the brightness. If building the firmware yourself, playing around with the button debouce settings could be used. 

```
    // Buttons
    #define BUTTON_MQTT_SEND_ALL_EVENTS     1
    #define BUTTON1_PIN  0     //USE GPIO0 for push button
    #define BUTTON1_CONFIG  BUTTON_PUSHBUTTON | BUTTON_SET_PULLUP | BUTTON_DEFAULT_BOOT
    #define BUTTON1_RELAY 1
    #define BUTTON1_PRESS BUTTON_ACTION_NONE
    #define BUTTON1_CLICK  BUTTON_ACTION_NONE
    #define BUTTON1_DBLCLICK  BUTTON_ACTION_DIM_DOWN
    #define BUTTON1_TRIPLECLICK  BUTTON_ACTION_DIM_UP
    #define BUTTON1_LNGCLICK  BUTTON_ACTION_TOGGLE
    #define BUTTON1_LNGLNGCLICK  BUTTON_ACTION_TOGGLE
```