xoseperez/espurna #2424: (bugs) tuya dimmer issues

timmeh87, 2021-01-27:
This is a continuation of the discussion at #2414 about the tuya generic dimmer module. This is multiple issues at once. Some worse than others

1:
In order to get v15 to work with my hardware (https://www.aliexpress.com/item/33026918862.html), I had to make a build of espurna that has serial debugging disabled. Im not positive if TUYA_GENERIC_DIMMER is aimed at this hardware specifically or not but it seems that it otherwise still uses UART0 to talk to the tuya dimmer so maybe this should always be disabled or moved to a different UART?

2:
On the last thread there are instructions that you have to follow in order to get espurna to recognize this dimmer hardware correctly (or something?) . Without this, there are two ON/OFF switches and two brightness sliders and it does not work correctly. Even with the fix, there are still two brightness sliders that seem to combine into a "total brightness". I think there must be a bug here since manual hacking is required to get a working dimmer, I just dont know exactly what it is. But I do think there should only be one brightness slider, although that is not the biggest inconvenience compared to the hacking.

3:
Whenever I use the web interface to set a value under 16, there is no change in the output of the dimmer, it remains "stuck" at the last value. I think there is a low-level issue sending a value lower than 16 to the Tuya module so the code does not send these values to the dimmer module, but I think it should still send 16, or 17, or whatever the minimum value is for more consistent behavior. For example if you change from 100 to 15 the output is still at 100. It would make more sense if it went to 17. Not sure if this affects the other interfaces like HTTP. Apparently enabling transitions can make this problem go away.

4
I am getting an intermittent crash when I send many brightness requests very quickly by using the arrow keys on the keyboard to move the slider on the web interface. The brightness gets stuck at some value (could be any value) and it wont respond to anything anymore until the dimmer device is rebooted (can reboot via web interface). A good temporary fix is "just dont do that" but since its a crash I think its bears mentioning

mcspr, 2021-01-27:
1 and 2 covered by https://github.com/xoseperez/espurna/commit/92d5e7becba23552c836bda8404305a8dc8eb07d
removing the need to configure stuff since the hw build flags set up a default to be configuration-only without the discovery step

Brightness + channel slider is a pretty old debate, but it perfectly represents what is actually happening. Setting brightness sets a specific 'ratio' for the channel, changing channel changes... channel value directly. Although, if have an idea for a patch, please do.

3 was already mentioned in the other issue, can you try changing the return line to  `rounded = 16;` so it goes along with the change but sets it to 16 when sending?

4 you might've seen crash decoding tools brief example in the issue template, but I will follow up on that anyway since it seems pretty easy to reproduce.

timmeh87, 2021-02-03:
I know you have made a few updates to the code base, very recent, but I did not take them yet. Im using code I got on january 24 via the "download repository" button. This version still requires the console hack to make the dimmer switch work correctly. I will attach the debug output at the end of the report as I think it is helpful.

**Bug description**
holding down the left arrow keyboard key on the channel #0 dimmer slider on the website, to move it from 190 to 20 causes the dimmer to become unresponsive for me, 100% of the time. only rebooting seems to fix it.

*A clear and concise description of what the bug is.*
it should not become unresponsive

**Steps to reproduce**
on my computer if i hold the arrow key on the "channel #0" slider, it moves down quickly and sends a dimming command each time, as seen on the debug console. After a few seconds the dimmer becomes unresponsive and "stuck" at a value, only rebooting fixes. The messages on the debug console change at this time and it looks like traffic on the serial bus stops maybe. Using the "Reboot" button on the website works.

**Device information**
----

[120952] [MAIN] ESPURNA 1.15.0-dev
[120952] [MAIN] xose.perez@gmail.com
[120952] [MAIN] http://tinkerman.cat

[120953] [MAIN] CPU chip ID: 0xEBEC2C
[120953] [MAIN] CPU frequency: 80 MHz
[120953] [MAIN] SDK version: 2.2.2-dev(38a443e)
[120953] [MAIN] Core version: 2.7.4
[120957] [MAIN] Core revision: 2843a5ac
[120957] [MAIN] Built: 2021-01-25 02:02:46
[120958] 
[120958] [MAIN] Flash chip ID: 0x1540C8
[120958] [MAIN] Flash speed: 40000000 Hz
[120958] [MAIN] Flash mode: DOUT
[120958] 
[120959] [MAIN] Flash size (CHIP)   :  2097152 bytes /  512 sectors (   0 to  511)
[120962] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[120963] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[120963] [MAIN] Firmware size       :   612624 bytes /  150 sectors (   1 to  150)
[120963] [MAIN] Max OTA size        :   409600 bytes /  100 sectors ( 151 to  250)
[120964] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[120964] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[120964] 
[120964] [MAIN] EEPROM sectors: 251, 250
[120967] [MAIN] EEPROM current: 250
[120967] 
[120968] [MAIN] EEPROM:  4096 bytes initially |   884 bytes used (21%) |  3212 bytes free (78%)
[120973] [MAIN] Heap  : 36608 bytes initially | 18672 bytes used (51%) | 17936 bytes free (48%)
[120974] [MAIN] Heap  : 16456 contiguous bytes available (9% fragmentation)
[120974] [MAIN] Stack :  4096 bytes initially |  1680 bytes used (41%) |  2416 bytes free (58%)
[120974] 
[120974] [MAIN] Boot version: 7
[120980] [MAIN] Boot mode: 1
[120981] [MAIN] Last reset reason: Reboot from web interface
[120981] 
[120981] [MAIN] Board: TUYA_GENERIC_DIMMER
[120982] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP RELAY SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[120982] [MAIN] OTA: ARDUINO ASYNCTCP WEB 
[120982] [MAIN] WebUI image: LIGHT
[120982] 
[121264] [MAIN] Firmware MD5: 39bb726c03e876af22657720fde00d2b
[121272] [MAIN] Power: 3480 mV
[121273] [MAIN] Power saving delay value: 10 ms
[121273] 

**Crash report**
I do not get a crash report on the website debug console, the website remains responsive in other ways (reboot button works) but the dimmer function ceases to work. 

**Tools used**
chrome 87, thinkpad p50, windows 10

**Additional context**
I built the binary myself but usually I can do this successfully and otherwise it all seems to work.

```

[... 248 lines cut ...]
[045532] [TUYA] =>: 55aa00000000ff
[045556] [TUYA] <=: 55aa000000010101
[054536] [TUYA] =>: 55aa00000000ff
[054560] [TUYA] <=: 55aa000000010101
[060006] [MAIN] System OK
[060466] [WEBSOCKET] Requested action: relay
[060467] [RELAY] #0 scheduled OFF in 0 ms
[060472] [RELAY] #0 set to OFF
[061445] [WEBSOCKET] Requested action: relay
[061446] [RELAY] #0 scheduled ON in 0 ms
[061451] [RELAY] #0 set to ON
[062455] [RELAY] Relay mask: 0b1
[063546] [TUYA] =>: 55aa00000000ff
[063571] [TUYA] <=: 55aa000000010101
[070296] [WEBSOCKET] Requested action: relay
[070297] [RELAY] #0 scheduled OFF in 0 ms
[070300] [RELAY] #0 set to OFF
[070769] [WEBSOCKET] Requested action: relay
[070769] [RELAY] #0 scheduled ON in 0 ms
[070774] [RELAY] #0 set to ON
[071778] [RELAY] Relay mask: 0b1
[072553] [TUYA] =>: 55aa00000000ff
[072577] [TUYA] <=: 55aa000000010101

```

timmeh87, 2021-02-03:
Also, I tried to build the latest code and TUYA_GENERIC_DIMMER wont build anymore. GOSUND_WP3 builds fine. I get this build output

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Tim\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Tim\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Tim\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M1S,led=2,sdk=nonosdk_191122,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600,float_support=disabled -ide-version=10812 -build-path C:\Users\Tim\AppData\Local\Temp\arduino_build_212926 -warnings=none -build-cache C:\Users\Tim\AppData\Local\Temp\arduino_cache_598024 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose C:\Users\Tim\Desktop\smarthome\smarthome\espurna\espurna\espurna.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Tim\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Tim\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Tim\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M1S,led=2,sdk=nonosdk_191122,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600,float_support=disabled -ide-version=10812 -build-path C:\Users\Tim\AppData\Local\Temp\arduino_build_212926 -warnings=none -build-cache C:\Users\Tim\AppData\Local\Temp\arduino_cache_598024 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mklittlefs.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -verbose C:\Users\Tim\Desktop\smarthome\smarthome\espurna\espurna\espurna.ino
Using board 'generic' from platform in folder: C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Using core 'esp8266' from platform in folder: C:\Users\Tim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Detecting libraries used...
"C:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Tim\\AppData\\Local\\Temp\\arduino_build_212926/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK22x_191122=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_GENERIC\"" -DLED_BUILTIN=2 -DFLASHMODE_DOUT [... cut ...] braries\\ESPAsyncTCP\\src" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\Hash\\src" "-IC:\\Users\\Tim\\Documents\\Arduino\\libraries\\xoseperez-fauxmoesp\\src" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\libraries\\ESP8266mDNS\\src" "-IC:\\Users\\Tim\\Documents\\Arduino\\libraries\\async-mqtt-client-0.8.1\\src" "-IC:\\Users\\Tim\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\

mcspr, 2021-02-03:
re. build error, either:
- select IDE build menu option mentioning lwip v2 higher bandwidth instead of lower memory
- set HOMEASSISTANT_SUPPORT build flag to 0 if you don't need it
- cut / comment out these lines:
https://github.com/xoseperez/espurna/blob/05cc081b7ddb69bae76a41d3de1fc9d147bb6202/code/espurna/homeassistant.cpp#L384-L388
(but HA is unlikely to work, if I remember the payload sizes correctly. may be wrong though)

> I do not get a crash report on the website debug console, the website remains responsive in other ways (reboot button works) but the dimmer function ceases to work.

That is interesting. Thanks for the report!

mcspr, 2021-02-06:
I very briefly tested the TUYA with the current dev version (after the https://github.com/xoseperez/espurna/commit/2f39d0db8a71533dac0cf7c27a719d0097a001d2, as noted just above)

Output should no longer ignore 0...16 values and instead it will make them into 16 and send that the serial. But, I wonder if that will cause the lights to go off...
Weird slider interaction should also be fixed, which was actually a giant bug in the way channels updated their values.