xoseperez/espurna #2428: Increase relays number

stefkamarton, 2021-03-18:
Hi!
I have a question about your code!

I have SonOff RF Bridge which is flash `espurna_1.14.1-itead-sonoff-rf-bridge.bin` and I want to connect to Home Assistant which is in Raspberry PI 4, I use MQTT protocol.
Unfortunatelly, this image supports just 7 relays.

Is there any way to increase this number e.g. 16 or 32?

Thanks!

mcspr, 2021-03-19:
rfbridge configuration sets `DUMMY_RELAY_COUNT` with the number of relays:
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/config/hardware.h#L765-L767
https://github.com/xoseperez/espurna/blob/62ad7da332f3f904ad8241a2b738be9820b196e4/code/espurna/relay.ino#L1426
Which is also overridable by the `relayDummy` setting (e.g. `set relayDummy 9` and `reset`, using webui debug panel or telnet)

Also note that the current dev branch another approach to managing rf codes via rpnlib scripting and without using relays
https://github.com/xoseperez/espurna/wiki/RPN-Rules#rfbridge
Development binary builds are broken atm though, I've not fixed that just yet

mcspr, 2022-01-12:
Not really applicable anymore. Also, RFB is no longer limited to the DUMMY provider setup as described above, and will work for any configured RELAY (virtual, or real, doesn't matter).
Plus, I'd strongly suggest to take a look at [RPNLib integration instead of using this as a relay proxy](https://github.com/xoseperez/espurna/wiki/RPN-Rules#rfbridge)

closed 2022-01-12 by mcspr