xoseperez/espurna #2432: ESP8266 - Button Switch on GPIO2

ddepilla, 2021-04-03:
Hello, i have a problem with ESP8266 and i need some help.
I'm trying to control the esp8266 with a wall switch to active a relay.
This is the connection of the esp8266:
GPIO 00: connected to a 10K resistor, this is connected to a LED and this is connected to GND
GPIO 02: connected to a Button switch and it's connected to GND.

I downloaded de 1.15.0 dev Espurna version.
I configured the hardware.h file:

**#elif defined(GENERIC_ESP01S_RELAY_V40)

    // Info
    #define MANUFACTURER        "GENERIC"
    #define DEVICE              "ESP01S_RELAY_40"

    // Relays
    #define RELAY1_PIN          0
    #ifndef RELAY1_TYPE
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL   // See #1504 and #1554
    #endif

    // LEDs
    //#define LED1_PIN            2
    //#define LED1_PIN_INVERSE    0

    // Buttons
    #define BUTTON1_PIN           2 // D2
    #define BUTTON1_CONFIG        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH

    #define BUTTON1_RELAY         1

    #define BUTTON1_PRESS         BUTTON_ACTION_TOGGLE
    #define BUTTON1_RELEASE       BUTTON_ACTION_TOGGLE**

I compiled this and uploaded to ESP8266.
All works fine except this case:

With the switch is enable (GPIO 02 connected to GND), i poweroff the ESP8266. When i PowerOn it again the ESP8266 startup in a strange mode, and still blocked (don't response the switch and the internet connection is lost)
With the switch enabled (GPIO 02 connected to GND), I poweroff the ESP8266. When I PowerOn back on, the ESP8266 boots into a strange mode and hangs (don't respond to the switch and internet connection is lost).

The only way to take control is set the switch off (GPIO 02 disconnected) and restart the ESP8266.

Someone can help me?
Thank you!

davebuk, 2021-04-03:
See: https://www.instructables.com/ESP8266-Using-GPIO0-GPIO2-as-inputs/

You can't use a wall/toggle switch on GPIO 02 if there is a chance it can be LOW at boot. You may be able to re-wire the switch as per the instructions in step 4 from above but I'm not sure if GPIO 0 is HIGH at boot and LOW after boot. @mcspr can you confirm either way?

With a bit of research on the net, you may find another GPIO that can be used on your board and use `BUTTON_DEFAULT_BOOT` in the button config.

ddepilla, 2021-04-03:
Thank you davebuk for your help.

To resolve it, i tried to use another gpio as input button switch. I tried to use GPIO 3, but it doesn't work ...
I used this configuration in hardware.h :

#elif defined(GENERIC_ESP01S_RELAY_V40)

    // Info
    #define MANUFACTURER        "GENERIC"
    #define DEVICE              "ESP01S_RELAY_40"

    // Relays
    #define RELAY1_PIN          0
    #ifndef RELAY1_TYPE
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL   // See #1504 and #1554
    #endif

    // Buttons
    #define BUTTON1_PIN           3 // D2
    #define BUTTON1_CONFIG        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
    //#define BUTTON1_CONFIG        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_LOW
    #define BUTTON1_RELAY         1

    #define BUTTON1_PRESS         BUTTON_ACTION_TOGGLE
    #define BUTTON1_RELEASE       BUTTON_ACTION_TOGGLE

    // LEDs
    #define LED1_PIN            2
    #define LED1_PIN_INVERSE    1
    #define LED1_MODE           LED_MODE_RELAY
    #define LED1_RELAY          1

The button switch has not effect on the led.
Do you know if i'm making some mistake?

Thank you again.
Best regards,
Damián.-

davebuk, 2021-04-04:
Is the actual physical relay changing state and the state of the relay in the webUI on each switch toggle? The debug screen should also show each time the relay changes.

Also, check the LED section the webUI that it's showing the same config as per your configuration and hasn't default to 'wifi connection state'.

ddepilla, 2021-04-04:
The relay change the status when i press the webUI's switch.
The debug screen show this.

[image]

[image]

The LED section show this:

[image]

The led is configured in the GPIO 2, in Mode "Switches Status" and related with Relay 0.

It works good.

The problem that i have is that I'm trying to connect a switch to the GPIO 3 (Rx of ESP8266) and the switch hasn't effect to LED and Relay.
The switch is connectd to GND, when i press it connect GPIO 3 (Rx) to GND.

Do you know what is wrong?

Thank you!
Damián.-

davebuk, 2021-04-04:
If using TX or RX pins as an input, you'll have to build adding:

`#define DEBUG_SERIAL_SUPPORT 0`

ddepilla, 2021-04-04:
Ok, i changed the configuration adding DEBUG_SERIAL_SUPPORT in disabled:

#elif defined(GENERIC_ESP01S_RELAY_V40)

// Info
#define MANUFACTURER        "GENERIC"
#define DEVICE              "ESP01S_RELAY_40"

#define DEBUG_SERIAL_SUPPORT    0

// Relays
#define RELAY1_PIN          0
#ifndef RELAY1_TYPE
#define RELAY1_TYPE         RELAY_TYPE_NORMAL   // See #1504 and #1554
#endif

// Buttons
#define BUTTON1_PIN           3 // D2
#define BUTTON1_CONFIG        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
//#define BUTTON1_CONFIG        BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_LOW
#define BUTTON1_RELAY         1

#define BUTTON1_PRESS         BUTTON_ACTION_TOGGLE
#define BUTTON1_RELEASE       BUTTON_ACTION_TOGGLE

// LEDs
#define LED1_PIN            2
#define LED1_PIN_INVERSE    1
#define LED1_MODE           LED_MODE_RELAY
#define LED1_RELAY          1

I had the same effect.
The switch connectd to GPIO 3 (Rx of ESP8266) doesn't work.

I verified the arduino.h file and the line "DEBUG_SERIAL_SUPPORT" is commented:

//--------------------------------------------------------------------------------
// Features (values below are non-default values)
//--------------------------------------------------------------------------------

//#define ALEXA_SUPPORT          0
#define API_SUPPORT            1
//#define BROKER_SUPPORT         0
#define BUTTON_SUPPORT         1
//#define DEBUG_SERIAL_SUPPORT   0
#define DEBUG_TELNET_SUPPORT   1
//#define DEBUG_UDP_SUPPORT      1
#define DEBUG_WEB_SUPPORT      1
#define DOMOTICZ_SUPPORT       1
//#define ENCODER_SUPPORT        1
#define HOMEASSISTANT_SUPPORT  1
//#define I2C_SUPPORT            1
//#define INFLUXDB_SUPPORT       1
//#define IR_SUPPORT             1
#define LED_SUPPORT            1
//#define LLMNR_SUPPORT          1
//#define MDNS_SERVER_SUPPORT    0
#define MQTT_SUPPORT           1
//#define NETBIOS_SUPPORT        1
//#define NOFUSS_SUPPORT         1
#define NTP_SUPPORT            1
//#define OTA_ARDUINOOTA_SUPPORT 1
//#define RFM69_SUPPORT          1
//#define RFB_SUPPORT            1
//#define RPN_RULES_SUPPORT      0
#define SCHEDULER_SUPPORT      1
//#define SPIFFS_SUPPORT         1
//#define SSDP_SUPPORT           1
#define TELNET_SUPPORT         1
//#define TERMINAL_SUPPORT       0
//#define THINGSPEAK_SUPPORT     0
//#define TUYA_SUPPORT           0
#define UART_MQTT_SUPPORT      1
#define WEB_SUPPORT            1

I tried withoit the Led configuration, due to both are referencing to the same Ralay (if there is possible confict), and i had the same result.

I tried with factory reset (if there was an old configuration), and nothing...

Any other idea?

Thank you.
Damián.-

davebuk, 2021-04-05:
This is my buttons, relay and led config for a sonoff basic R1 with toggle switch on GPIO 14. The built in push button is set as button 1 and I've disabled some of its ACTIONS. Button 2 is then setup for the toggle switch.

Can you check the voltage on Pin 3 is changing between 3v and 0v when you change over the switch state? 

I've done testing before with a small length of solid core wire held to a ground point, then carefully touched it to the GPIO I've configured. That will pulse the GPIO from pulled up HIGH to ground. 

**Remember there are mains voltages on the board!**

```
//My config
    #define DEBUG_SERIAL_SUPPORT    0 // If using TX or Rx as input
    

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_CONFIG      BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY       1
    #define BUTTON1_PRESS       BUTTON_ACTION_NONE
    #define BUTTON1_CLICK       BUTTON_ACTION_TOGGLE
    #define BUTTON1_DBLCLICK    BUTTON_ACTION_NONE
    #define BUTTON1_LNGCLICK    BUTTON_ACTION_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_ACTION_NONE
    #define BUTTON2_PIN         14       
    #define BUTTON2_CONFIG      BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_BOOT
    #define BUTTON2_RELAY       1
    #define BUTTON2_PRESS    BUTTON_ACTION_TOGGLE
    #define BUTTON2_RELEASE  BUTTON_ACTION_TOGGLE

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1
```

ddepilla, 2021-04-06:
Hello Dave, my configuration is so similar to your description... with the difference of the HW, I have de ESP8266 on the Relay ESP 01s.
My ESP8266 have 8 pins, with 4 GPIO (0 to 3). The GPIO 1 and 3 are Tx and Rx, so i understand that is important to disable "DEBUG_SERIAL_SUPPORT" setting to 0.

I can confirm that the GPIO 3 (RX) is connected to GND when i press the botton, because i have connected a tester. 
The voltage is in 3,5v when the switch is off, and is in 0v when the switch is on, but it hasn't effect to the relay number 1.

I can't understand what is wrong...

Is possible that there is a bug in this version?

Any other idea?

Thank you so much!
Best regards,
Damián.-

davebuk, 2021-04-06:
I'm starting to get to the limit of my knowledge I'm afraid. I suggest you do a search for building with `custom.h`. See https://github.com/xoseperez/espurna/wiki/PlatformIO#customize-build-settings for details. I find it easier to develop custom builds leaving the core files alone and allows for easy updating of the core espurna files.

The other thing to check is in the webUI debug screen. Type `gpio` and press send and you'll get the config/status of the GPIOs. If you then change the state of one and re-run `gpio` you should see the new state. That way you know espurna is reading the GPIO logic. If the GPIO state doesn't change, you'll know there is an issue with the build or the way the firmware is using the GPIO signals.

ddepilla, 2021-05-23:
Hello Dave, after some weeks i was able to return to this topic.

I debug it, and i confirm than the input doesn't change when i put the GPIO1 (TX) or GPIO3 (RX) to Low or High...
Always the GPIO1 and GPIO3 are LOW (before and after made the change):

[image]

So, this is the explication of why it don't change the Led status...

Anyway, i don't understand why this inputs don't works... Is like the "#define DEBUG_SERIAL_SUPPORT 0" doesn't take effect...

Any idea?
Can i report it like a Bug?

Other question, do you know why the GPIO1 has an asterisk (*) in the debug screen?:

[image]

And the others GPIO's haven't got it?

Thank you.
Best Regards,
Damián.-

mcspr, 2021-05-24:
> I debug it, and i confirm than the input doesn't change when i put the GPIO1 (TX) or GPIO3 (RX) to Low or High...
Always the GPIO1 and GPIO3 are LOW (before and after made the change):

Reading the config:
```cpp
#define BUTTON1_PIN 3 // D2
#define BUTTON1_CONFIG BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
```
We expect D2 to be pulled to GND to trigger the button press event. Otherwise, it should be pulled up to 3v3 to stay in the default state. Config sets internal pullup, have you tried doing this externally?

> Other question, do you know why the GPIO1 has an asterisk (*) in the debug screen?:

Means something has 'locked' the GPIO, e.g. `button` command would show the gpio3 as being used

davebuk, 2021-05-24:
I used to use the TX/Rx pins on Sonoff basic boards as an input but had issues once where it just wouldn't work. As I don't need the status LED, I removed it from the circuit and used that GPIO instead, removing the configs for LEDs.

Can you find out if there are any other GPIOs available on the board? You could configure multiple buttons  `BUTTON1` `BUTTON2` `BUTTON3` etc for different GPIOs and see if any of them work?

closed 2022-05-05 by mcspr