xoseperez/espurna #2433: Factory reset seemed to crash device.

davebuk, 2021-04-03:
@mcspr 
Hi

Can you check what happens when the factory reset code is run? Ref the comment below, after carrying out a factory reset from the webUI I lost access. Even though the built in AP did appear to connect to, I wasn't able to re-configure or get to the webUI. Re-flashing via serial was the only way to fix it.

> I decided to do a factory reset from the webUI to clear out all settings and start again. In doing so I now can't get to the web interface! I can connect to the espurna AP using the default password but the webUI on 192.168.4.1 doesn't open. I may have to look at reflashing using the hardware pins.

_Originally posted by @davebuk in https://github.com/xoseperez/espurna/issues/2422#issuecomment-785400377_

I have just tried https://github.com/xoseperez/espurna/commit/cb23c9444b16cc5e254f34a69898a5aff6528698 on a HNC BNC50 wifi switch and although it initially connected I lost connection. I didn't have the relay showing in the webUI so decided to factory reset the device. This time I couldn't access the built in wifi AP at all. Physical button didn't switch the relay and the LED is hardly lit at all. I've checked the 240VAC and 3.3VDC so don't think the device has electrically failed.

I have re-flashed a basic core-webUI .bin and a known previously working .bin I used on this device. The USB to serial connects and says it has loaded the firmware, but the built in AP never appears.

I tried opening a putty serial connection but it doesn't connect. Could the memory chip have randomly failed or could the factory reset have caused an issue?

mcspr, 2021-04-05:
Factory reset itself is pretty basic. We keep settings as a raw 4KiB blob in ram, a erasing it's only a matter of writing 0xff into it and saving to the flash
https://github.com/xoseperez/espurna/blob/4d157ccd5bd5ffefa8b0bca79c4b2196c8a3e5dc/code/espurna/storage_eeprom.h#L54-L58

And there's not even a boot log from the esp8266 itself? Monitor with 74880 speed should still show something like this
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9

csum 0xc9
v0008c210
~ld
```
It could be something with the hardware, but I'd check this first to see if the fw is loading at all.

davebuk, 2021-04-05:
@ 74880 and 9600 I get the following :-) Every few seconds putty gets another set of random characters.
```
@▒Z-nP▒M
%▒ͅ      %Y+Q
                ▒E
                  xÈ▒▒RKT\%     )▒@▒▒NRA▒       ڻ^
                                                        J▒
                                                                        ▒Z▒▒▒▒Q▒                           5@9H▒▒▒
ě▒+▒▒@J▒1@▒P▒J޵iX*▒@J@1@Ā▒▒▒VXK▒Ḇ(q▒q▒&'▒▒Mo
                                            ▒rCX?       ▒9@^▒P▒M▒
                                                                 ▒D!RPX1▒▒q▒▒z
▒E      ▒▒▒
▒
▒3▒▒ZA  I
▒?kC▒▒5@▒9▒▒R▒▒▒'LB▒=%▒փU@.▒MB9A
▒▒y▒
5▒▒a▒▒▒ 1+▒5@1▒R▒▒▒▒▒▒▒▒▒▒▒@H▒R▒M
                                 1C@▒1▒q▒C/▒@▒9▒▒▒▒▒▒▒
        ▒▒V!▒LM▒▒
ˢL
▒CMi
    !̃
▒R^▒P▒1▒ZPM+▒▒= *▒▒
▒y▒L
▒▒@▒PX▒Y
ˢL
▒▒▒▒@KLNJ▒▒▒▒▒@J
%▒T▒    ▒y▒     9E▒▒
)CLC▒a▒P▒%▒֋݅▒▒▒ N
             9▒▒▒݀p▒
▒@▒@▒q▒V▒▒\▒▒C▒▒▒@J@1▒q▒Y▒XXP▒▒@▒
/C▒P▒▒▒▒Q-1▒▒▒[C▒▒▒@J@1▒A▒P▒▒▒@\▒1▒
 ▒9▒▒;?mN%▒
          ˉ
           ▒▒;? ▒3@▒Ѝ#▒%▒▒D2▒▒▒@D!▒▒ʤ▒
!,▒▒
1L▒ĀDqB▒7▒*▒Ƣ▒
▒▒▒@J@1▒T2B+I▒X7▒
                 ▒▒▒*▒▒▒q▒1▒▒▒▒C▒▒▒s▒7oR7▒▒▒[▒+LC▒▒P@▒X▒
                                                        X▒
XP▒▒@̊▒PA▒TI▒▒Q(^▒P▒C▒▒▒@▒ID#Ds▒▒V▒V▒▒\z
                                       1k
▒▒▒▒▒▒LP▒▒      5NH
#▒KJ▒▒A▒qaP▒E@▒▒▒▒,
7▒VI▒C▒▒▒@▒LL'I▒q▒▒@▒▒▒▒
▒▒▒1B▒+Z▒N▒?    ▒9▒▒V▒;!))5L▒7▒▒v▒N▒    ▒9▒▒▒▒Z▒▒▒7+▒▒▒@▒▒гZV▒q▒T▒▒'▒▒▒
```

davebuk, 2021-04-05:
Fixed.

I have been using 'NodeMCU-PyFlasher-4.0-x64' to flash, but I went back and tried 'ESPEasy flash tool'. This allowed me to get the device working again. :-) Strange!

mcspr, 2021-04-05:
Only 74880 matters, and only on boot since further reboots switch into 115200 set by us. i.e.
- connect uart<->usb with the computer
- connect wires to the device but don't power on
- open up monitor, power on

Good that it is not completely broken.

This got me a bit suspicious about the WebUI OTA though. With something like ArduinoOTA or NOFUSS, it sends the file with a md5 hash of contents (aka signature, ensure file is the same on both ends), so esp8266 can compare the result when written to the internal flash. In theory it is possible to write some garbage to some of the sectors. e.g. loose strings, or some functions.
Will need to look into it a bit more though, and js does not really have hashing algorithms built-in or requires the site that is using it to be https (...which is whole other issue...) :/

closed 2021-04-05 by mcspr

davebuk, 2021-04-05:
As the device was not booting correctly after trying the WiFi reconnect code additions, like you said, maybe something got corrupted when I did the factory reset. Anyway, thanks for the help. I'll get back to testing the latest code over the next few days.