xoseperez/espurna #2434: webUI not showing current wifi `keys` values.

davebuk, 2021-04-03:
I'm just trying https://github.com/xoseperez/espurna/commit/cb23c9444b16cc5e254f34a69898a5aff6528698 on a wemos d1 mini. I've set up the device and the wifi re-connect seems to work in my initial testing.

I've set a static IP and my wifi credentials while connected to the built in AP and restarted. The device connects fine to my wifi but the previously saved credentials are not shown in the wifi tab in the webUI. All credentials are listed in `keys`, just not in the webUI.

mcspr, 2021-04-04:
Should be fixed after currently at 0ee55ba5c001105ca6daa6cb1103e27d3cd462c8?
(after https://github.com/xoseperez/espurna/commit/da5bc889169e879553ba308f1e6a8c3cca5b7e2c and https://github.com/xoseperez/espurna/commit/7226a7d37f1c723898fb55b5fbc8617566712f47)

davebuk, 2021-04-04:
Fixed.

closed 2021-04-04 by davebuk