xoseperez/espurna #2435: PlatformIO VSCode build error, firmware.map: No such file or directory.

davebuk, 2021-04-10:
I've suddenly started having build failures in PlatformIO. The last part of the build process is shown below. I've tried building three different environments and they all fail with the same error.

I've deleted the `\.pio` and tried again, but no good.

Could this be an espurna issue or PlatformIO?
```
Archiving .pio\build\espurna-core-webui-1MB\libFrameworkArduino.a
Linking .pio\build\espurna-core-webui-1MB\firmware.elf
xtensa-lx106-elf-g++: error: Software\Firmware: No such file or directory
xtensa-lx106-elf-g++: error: Builds\Espurna\dev\espurna\code\.pio\build\espurna-core-webui-1MB/firmware.map: No such file or directory*** [.pio\build\espurna-core-webui-1MB\firmware.elf] Error 1
==================================================== [FAILED] Took 108.99 seconds ====================================================
Environment             Status    Duration
----------------------  --------  ------------
espurna-core-webui-1MB  FAILED    00:01:48.994
=============================================== 1 failed, 0 succeeded in 00:01:48.994 =============================================== 
The terminal process "C:\Users\Server\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'espurna-core-webui-1MB'" terminated with exit code: 1.
```

mcspr, 2021-04-11:
Looks like there are spaces in the path. Does this patch work?
```diff
diff --git a/code/scripts/pio_main.py b/code/scripts/pio_main.py
index 9e1ae58..a995616 100644
--- a/code/scripts/pio_main.py
+++ b/code/scripts/pio_main.py
@@ -26,7 +26,7 @@ import os

 CI = "true" == os.environ.get("CI")

-env.ProcessFlags("-Wl,-Map -Wl,${BUILD_DIR}/${PROGNAME}.map")
+env.ProcessFlags("-Wl,-Map -Wl,\\\"${BUILD_DIR}/${PROGNAME}.map\\\"")

 # Always show warnings for project code
 projenv.ProcessUnFlags("-w")
```

davebuk, 2021-04-11:
Yes, that works.

commit 1ed00f57 (2021-04-12): pio: make sure the .map path is properly quoted

fix #2435
closed 2021-04-12 by mcspr