xoseperez/espurna #2437: No plug (SP111) control after flashing espurna.

Lookzord, 2021-04-15:
Recently I bought few Gosund SP111 plugs and wanted to flash them to espurna - to integrate them into HA.
I followed a tutorial where you can flash this plug over the WIFI using tuya-convert. After strictly following everything Pi4 console said that espurna is flashed successfully. But right after I logged into plugs ESPURNA-XXXX SSID and got onto webserver I cannot see any switches or power monitors, mqqt configs, nothing at all. 
[image]
What can I do now? Did I kill the plug?
Did I make a mistake and should have flashed tasmota - and then espurna?

mcspr, 2021-04-15:
This is a base image, note the device name. Either via WebUI OTA or espota.py you can flash the full .bin

Lookzord, 2021-04-15:
Could you please tell me how to do it? espota.py requires wired connection to the plug i suppose? WebUI seems like a way to go but how can I do it?

Lookzord, 2021-04-15:
Ok I just found this thread https://github.com/xoseperez/espurna/issues/1998 and decided to upgrade bin file through WebUI using suggested BLITZWOLF_BWSHPX. But after upgrading my plug is now blinking every 10 seconds and does not make AP anymore. Well I think I just locked myself out of this plug.

mcspr, 2021-04-16:
Does it toggle the switch after initial power on?
Also note that it also updated the AP password if it was changed in the webui

Lookzord, 2021-04-16:
Nah, the plug went to live on its own. After following some documentation and tutorials I found out how to flash espurna properly on my device. As for now I had to wire flash tasmota with tasmotizer and after proper configuration the plug works like a charm again. The next step for me is to prepare right esp config file in home asistant and ota flash it from webui. 
Hope It will do the job.
Thanks for your insight! :)

mcspr, 2021-04-16:
Flashing led was at least a sign that something worked, but no AP probably meant it crashed immediately :/

I remember some issue or gitter mention of the similar situation... was the plug used at all with the stock fw, or joined the wifi network while using the tuya app? and do you happen to have a backup image from tuya-convert?
Also note if tasmota was connected to a wifi network, there might be a similar bug after OTA that was (hopefully fully) fixed in versions after https://github.com/xoseperez/espurna/commit/0586f5d4cc631aef0ab366cf045131f9aaa940e7 (27 Sep 2020)

closed 2021-04-16 by mcspr