xoseperez/espurna #2441: Only dummy switches 0-9 are declared to Home assistant discovery

Sag06, 2021-05-04:
Current dev here.
I successfully increased Itead RF Bridge dummy relays to 19. All is fine on ESPURNA web UI. Even Alexa discovered them flawlessly.
Problem is that Home assistant discovery can only discover 0 to 9 switches.
It seems Espurna can only declare one digit number switches.

Note: I use dummy switches to execute HA automations with Alexa Hue emulation.

mcspr, 2021-05-17:
idk what is this about, I am seeing all 20 switch entities I added via `set relayDummy 20` after discovery. All MQTT messages are sent looking at the logs as well (homeassistant/config... (PID 1)) + mosquitto_sub. Unless there is another issue with retries, you are sure that you are running the latest commits?

Sag06, 2021-05-21:
You pointed me to solution. On ESPURNA UI, I disabled HA discovery, enabled again and Voilà, all 19 dummy switches are there!
Not sure why, but they weren't discovered before in spite of ESPURNA and HA rebootings.  Thanks indeed.

closed 2021-05-21 by Sag06

mcspr, 2021-05-22:
Could you try a couple of times and look at the our log? MQTT messages *could* fail in the moment (seen as PID 0 / same message sent twice), but there should be retries scheduled immediately after.
Btw, do you build with lwip2 low mem or high bandwidth?

Also note 71ddf35022678667d0269ecc9c60c69bdab68079 that fixes possible issues on startup with 16+ relays