xoseperez/espurna #2444: Gosund WP2

mcspr, 2021-05-06:
Just some information that may help others...
1) I am not using a SHP7, but a Gosund WP2, which I found out to be quite similar to each other, but still, read this with care;
2) I was able to program my WP2s using tuya-convert 2.4.3 and OTA -- out of the box.  But be careful, the manufacturing date has huge impact on your success (newer firmwares don't seem to be tuya compatible at all -- reportedly 2.4.3, 2.4.4, or 2.4.5);
3) after uploading my own code and experimenting with the GPIOs (the documented numbers were not working), I lost OTA and had to crack the plug open.  I connected the wires (Tx, Rx, Vcc, GND, GPIO0) and after a few trials (watch for Vcc=3.3V, if instead it is close or lower than 3.1V, it won't program) and was able to 'play' more with the GPIOs; 
4) some configurations of the GPIOs as 'input' or 'output' will prevent the device from communicating with the outside (e.g. UART), and even turn off or disturb the WiFi to the point it won't work anymore.
I am still not able to determine the function of all GPIOs (as relay1, or relay2, etc...)  If any one knows those for sure, I would appreciate any hints.

_Originally posted by @GNDeSouza in https://github.com/xoseperez/espurna/issues/2167#issuecomment-831368142_

mcspr, 2021-05-06:
> Yeah... Those don't work with my version of WP2 (V1.1 2018) and the simple program I wrote to test the GPIOs. For one, there are four LEDs and not just 2 (two red and two blue). Also, some GPIOs affect the brightness of those same LEDs.
I am collecting more information and will post here (and at Tamota) once I have it all listed.

Per https://templates.blakadder.com/gosund_WP2.html there are 3 pins controlling things related to LED - gpio1, gpio2 and gpio13. Using gpio1 implies disabling serial debug and terminal input, default build will try to use those for Serial reads / writes which will cause a conflict. Also note that you should not init gpio6, gpio7, gpio8 and gpio11 as those are technically reserved the drive the spi flash connection

GNDeSouza, 2021-05-07:
Hi @mcspr!  I guess I was indeed messing up with the SPI.  
Now, do you have any idea why/how the device (i.e. WP2 -- not sure about SHP7) would 'latch' the input to zero? 

That is, I measured with oscilloscope/voltmeter, and I noticed that when the device is powered up, both inputs (GPIO0 and GPIO4) read logic 1, but if I press the buttons connected to those inputs, they read logic 0 and won't go back to logic 1, unless I power cycle the device.

The device is running my own program, which only sets the modes of the GPIO ports to input or output according to the template you cited above (from TASMOTA).  But again, if I measure the voltage on, say, GPIO0 at the beginning, it is logic 1, but after I press button connected to that GPIO0 (button1), it won't read logic 1 unless I power cycle the device.  Any idea why?

mcspr, 2021-05-07:
`INPUT_PULLUP`?

GNDeSouza, 2021-05-07:
Yeah... that's what I mean by parasitic capacitor.  But then again, how would one "reset" the input back to 1 without having to power cycle it -- and I had TASMOTA loaded and it worked just fine, so it must be programming some other GPIO or doing something to GPIOs 0 and 4 to bring them back to logic 1 after we press the buttons, right?

mcspr, 2021-05-07:
:oops: I did not type the rest.

This was mentioned on gitter - https://gitter.im/tinkerman-cat/espurna?at=5df8a3a20dc628523e4de80c
Since for some reason newer plugs don't have hw pullups anymore

Tasmota tries to set pullup for buttons, unless the pin type is `Button#n` (see gpio conversion page):
https://github.com/arendst/Tasmota/blob/3c4acc952868bcb1109bf9d756dd43b5d43f63dc/tasmota/support_button.ino#L97

GNDeSouza, 2021-05-07:
Thanks!  I was not aware of the INPUT_PULLUP option in the ESP8285.   I should have thought about that.   ;-)