xoseperez/espurna #2448: Schedule TAB

etardioli, 2021-05-14:
Hi, im really really new in Espurna, im trying to replace ESPeasy with some Relay and DHT22.
Why im not seeing the schedule Tab?
[image]
Thanks!

mcspr, 2021-05-16:
This could be reflected in the web developer tools console as some kind of error?
Does this work on nightly builds? 1.14.1 vs current dev diverged a lot, and there were some long standing issues exactly like this

etardioli, 2021-05-17:
Hi! thks for your time,  to enable schedule maybe needs to be compiled to use GPIO? I try several versions, 1.14.0 - 1.13.5 and none of them has the schedule tab

mcspr, 2021-05-18:
You should see SCHEDULER in the support: ... list on boot / after calling `info` command.
If you are building yourself, note that old versions building with Core 2.3.0 will experience https://github.com/xoseperez/espurna/issues/1610 and https://github.com/xoseperez/espurna/issues/1614 (may or may not be related though). Or simply loose content b/c of low memory conditions, which is also fixed in development versions

etardioli, 2021-05-18:
Im not building by myself, the version i've is for wemos d1 mini.

etardioli, 2021-05-19:
[image]
Yes, i tried. Is a step i missing?

etardioli, 2021-05-19:
support: ALEXA API BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS MQTT NTP ARDUINO_OTA OTA_CLIENT RELAY SCHEDULER TELNET TERMINAL THINGSPEAK WEB

etardioli, 2021-05-19:
[image]
inspect the gui as u suggested and i think those items are for the Schedule tab,  but is not visible in any browser or device...

etardioli, 2021-05-19:
[image]

mcspr, 2021-05-19:
There are no relays though:
https://github.com/xoseperez/espurna/blob/3078ad16c7db37f4b4b1b839d87076091dd5c6ef/code/espurna/scheduler.cpp#L153-L156
Panels prefixed with `sch` will be shown only when there is at least one.

e.g. `set relayGpio0 5` to have one on GPIO5 or `set relayDummy 1` for a virtual one

etardioli, 2021-05-19:
[image]
Thank you very much!

closed 2021-05-19 by mcspr