xoseperez/espurna #2449: Temperature Sensor AddOn for Shelly 1/1PM

andreaconfa, 2021-05-14:
Is the Temperature Sensor AddOn for Shelly 1/1PM supported?

mcspr, 2021-05-17:
It is for 1PM, per https://github.com/xoseperez/espurna/blob/8892122091ce1c3282ebb65f9e88be90a284ce4f/code/espurna/config/hardware.h#L3755

andreaconfa, 2021-05-18:
> It is for 1PM, per
> 
> https://github.com/xoseperez/espurna/blob/8892122091ce1c3282ebb65f9e88be90a284ce4f/code/espurna/config/hardware.h#L3755

is possible to implement it with the shelly 1 standard?

mcspr, 2021-05-19:
Don't think so, as there no hardware support? Looking at https://templates.blakadder.com/shelly_1.html, external pins should not be used for any peripherals

andreaconfa, 2021-05-19:
it's supported... external pins is used in all the shelly devices to program the flash and also for the temperature sensor

andreaconfa, 2021-05-19:
> Don't think so, as there no hardware support? Looking at https://templates.blakadder.com/shelly_1.html, external pins should not be used for sensors

it's supported... external pins is used in all the shelly devices to program the flash and also for the temperature sensor

mcspr, 2021-05-19:
Can you link to the description of the sensor that you want to use?

andreaconfa, 2021-05-19:
> Can you link to the description of the sensor that you want to use?

The sensor is this one (it's an original sensor from shelly): https://shop.shelly.cloud/temperature-sensor-addon-for-shelly-1-1pm-wifi-smart-home-automation#312
As you can see is compatible with Shelly 1, Shelly 1PM and DS18B20

mcspr, 2021-05-19:
Looks like Shelly made this incompatible with [DS18x20 sensor code we already have](https://github.com/xoseperez/espurna/blob/dev/code/espurna/sensors/DallasSensor.h), as it uses 2 wires instead of one :/
Looking at the implementation here, it seems to be the only way to talk to it:
- https://github.com/arendst/Tasmota/blob/1966dbbe412107a65c87aaa4d392ee5d043728fb/tasmota/xsns_05_ds18x20.ino
- https://templates.blakadder.com/shelly_temperature_sensor.html (DSB_OUT pin -> Output, DS18x20 pin somehow work as input-only)

andreaconfa, 2021-05-19:
The sensor itself have only 3 wires. 2 for power supply and 1 for the data

Inviato da iPhone

  *   https://github.com/arendst/Tasmota/blob/1966dbbe412107a65c87aaa4d392ee5d043728fb/tasmota/xsns_05_ds18x20.ino

andreaconfa, 2021-05-19:
> Looks like Shelly made this incompatible with [DS18x20 sensor code we already have](https://github.com/xoseperez/espurna/blob/dev/code/espurna/sensors/DallasSensor.h), as it uses 2 wires instead of [...]

is the implementation in espurna possible? i really like this project (in fact i'm using it on 40+ devices in my home) and it works really well compared to tasmota

andreaconfa, 2021-05-19:
> It is possible, ofc. Just means we would need to merge OneWire library with Dallas sensor code and name it something like ShellyTemperature.

That would be great