xoseperez/espurna #2451: H801 crashes when changing colors using drag

luebbe, 2021-05-17:
**Bug description**
The device (H801) reboots when the color is changed by **dragging** the brightness/saturation selector with the mouse over the square or by **dragging** the color selector. It does not crash when you just **click** on the desired color or brightness/saturation.

My guess is that there is a problem, when a color transition is triggered while there is still one in progress
Therefore I also added the device to Home Assistant in order to see if the crash also occurs when the color selector in the HA UI is dragged and the commands are received via MQTT. But in this case it works fine. No crashes.
The crash only occurs, when the selector is dragged on one of the two native Espurna color selector UI controls.

**This is the log output**

I started dragging after the `[026028] [RELAY] #0 set to ON` line.

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,3)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
[000104] [MAIN] System OK
[000107] [MAIN] ESPURNA 1.15.0-dev.git88921220 built 2021-05-16 20:40:57
[000107] [MAIN] xose.perez@gmail.com
[000107] [MAIN] http://tinkerman.cat
[000109] [MAIN] CPU chip ID: 600194422440
[000112] [MAIN] SDK: 2.2.2-dev(38a443e)
[000116] [MAIN] Arduino Core: 2.7.4
[000119] [MAIN] Support: ALEXA API BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS MQTT NTP ARDUINO_OTA OTA_CLIENT RELAY SCHEDULER TELNET TERMINAL THINGSPEAK WEB
[000145] [TELNET] Listening on port 23
[000147] [WEBSERVER] Webserver running on port 80
[000150] [LIGHT] Provider: DIMMER
[000153] [LIGHT] Number of channels: 5
[000173] [RELAY] Number of relays: 1, boot mask: 0b1
[000174] [RELAY] #0 set to OFF
[000177] [BUTTON] Number of buttons: 0
[000180] [LED] Number of leds: 1
[000182] [MQTT] AsyncMqttClient, SSL DISABLED, Autoconnect ENABLED, Buffer size 1024 bytes
[000184] [MQTT] Client DISABLED, DISCONNECTED
[000211] [NTP] Startup delay: 3 (s), Update delay: 2266 (s)
[000230] [NTP] Server: 192.168.0.1, TZ: CET-1CEST,M3.5.0,M10.5.0/3
[000236] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000319] [MAIN] Uptime: 00y 00d 00h 00m 00s
[000320] [MAIN] 24224 / 38928 bytes available (24216 contiguous)
[000321] [MAIN] VCC: 3523 mV
[000322] [WIFI] Initial
[000338] [WIFI] State Boot -> Idle
[000340] [WIFI] Mode changed to STA
[000351] [WIFI] State Idle -> Init
[000366] [WIFI] State Init -> Connect
[000368] [WIFI] Connecting
[000378] [WIFI] State Connect -> WaitConnected
[001174] [RELAY] Relay mask: 0b0
[004177] [WIFI] State WaitConnected -> Connected
[004178] [MQTT] MQTT brokers found: 0
[004181] [MDNS] OK
[004182] [WIFI] Connected to BSSID 2C:3A:FD:FE:7D:89 SSID onklan
[004193] [WIFI] State Connected -> Idle
[005004] [MQTT] Connecting to broker at 192.168.0.10:1883
[005004] [MQTT] Client ID: ESPURNA-422440
[005004] [MQTT] QoS: 0
[005004] [MQTT] Retain flag: Y
[005006] [MQTT] Keepalive time: 300 (s)
[005010] [MQTT] Will topic: h801-test/status
[005035] [MQTT] Subscribing to h801-test/light_json/set (PID 1)
[005036] [MQTT] Subscribing to h801-test/action/set (PID 2)
[005036] [MQTT] Subscribing to h801-test/led/+/set (PID 3)
[005041] [MQTT] Subscribing to h801-test/relay/+/set (PID 4)
[005046] [MQTT] Subscribing to h801-test/pulse/+/set (PID 5)
[005053] [MQTT] Subscribing to h801-test/brightness/set (PID 6)
[005058] [MQTT] Subscribing to h801-test/rgb/set (PID 7)
[... 155 lines cut ...]
[008140] [MQTT] Sending h801-test/board => HUACANXING_H801 (PID 1)
[008146] [MQTT] Sending h801-test/host => h801-test (PID 1)
[008152] [MQTT] Sending h801-test/desc => RGB Controller Dachstuhl oben (PID 1)
[008158] [MQTT] Sending h801-test/ssid => onklan (PID 1)
[008164] [MQTT] Sending h801-test/ip => 192.168.0.43 (PID 1)
[008169] [MQTT] Sending h801-test/mac => 60:01:94:42:24:40 (PID 1)
[008175] [MQTT] Sending h801-test/rssi => -37 (PID 1)
[008180] [MQTT] Sending h801-test/uptime => 8 (PID 1)
[008185] [MQTT] Sending h801-test/datetime => 2021-05-17 19:29:08 (PID 1)
[008192] [MQTT] Sending h801-test/freeheap => 18552 (PID 1)
[008197] [MQTT] Sending h801-test/loadavg => 0 (PID 1)
[008203] [MQTT] Sending h801-test/vcc => 3523 (PID 1)
[008207] [MQTT] Sending h801-test/light_json => {"state":"OFF"} (PID 1)
[008214] [MQTT] Sending h801-test/relay/0 => 0 (PID 1)
[008219] [MQTT] Sending h801-test/hex => #000000 (PID 1)
[008224] [MQTT] Sending h801-test/rgb => 0,0,0 (PID 1)
[008229] [MQTT] Sending h801-test/hsv => 0,0,0 (PID 1)
[008234] [MQTT] Sending h801-test/mired => 326 (PID 1)
[008239] [MQTT] Sending h801-test/channel/0 => 0 (PID 1)
[008244] [MQTT] Sending h801-test/channel/1 => 0 (PID 1)
[008249] [MQTT] Sending h801-test/channel/2 => 0 (PID 1)
[008255] [MQTT] Sending h801-test/channel/3 => 0 (PID 1)
[008260] [MQTT] Sending h801-test/channel/4 => 0 (PID 1)
[008265] [MQTT] Sending h801-test/brightness => 232 (PID 1)
```

mcspr, 2021-05-18:
> My guess is that there is a problem, when a color transition is triggered while there is still one in progress

Will need to look at this. My guess as well is something changing the object *while* it is used. There are some conditions that panic() when that sort of thing happens, but I kind of expected for those to produce a trace pointing at the culprit.

afaik HASS UI only generates an MQTT message when dragging stops? Our color widget does not do this, and sends all updates as they happen

closed 2021-07-23 by mcspr

mcspr, 2021-07-23:
I think this should solve it? Locking seems out-of-place right now, so is the manual panic(). I only briefly tested this, so please check it out with the current dev build