xoseperez/espurna #2452: It doesn't connect to a hidden wifi access point

GitJa21, 2021-06-06:
Actually, the question.
I have a few sonoff devices and a hidden WiFi hotspot. With your firmware, devices cannot connect to the router.

mcspr, 2021-06-07:
Does it work with WiFi scan disabled? See WIFI tab in the UI, or change `wifiScan` settings value to `0`

Does it work after modifying wifi.cpp like this?
```diff
diff --git a/code/espurna/wifi.cpp b/code/espurna/wifi.cpp
index 7c007e93..c7f7b165 100644
--- a/code/espurna/wifi.cpp
+++ b/code/espurna/wifi.cpp
@@ -946,12 +946,13 @@ bool start(Success&& success, Error&& error) {
     // https://github.com/espressif/ESP8266_NONOS_SDK/issues/103#issuecomment-383440370
     // Which could be useful if scanning needs to be more aggressive or switched into PASSIVE scan type

-    //scan_config config{};
-    //config.scan_type = WIFI_SCAN_TYPE_ACTIVE;
-    //config.scan_time.active.min = 100;
-    //config.scan_time.active.max = 300;
+    scan_config config{};
+    config.scan_type = WIFI_SCAN_TYPE_ACTIVE;
+    config.scan_time.active.min = 100;
+    config.scan_time.active.max = 300;
+    config.show_hidden = 1;

-    if (wifi_station_scan(nullptr, &internal::complete)) {
+    if (wifi_station_scan(&config, &internal::complete)) {
         internal::task = std::make_unique<Task>(std::move(success), std::move(error));
         return true;
     }
```

GitJa21, 2021-06-07:
Oh, thank you!
Its pretty good conected when scan is disabled.
Because of color gamma and old monitor I was not able to see that prase "Disable this option if you are connecting to a single access point (or router) or to a hidden SSID." Such a shame...

closed 2021-06-07 by GitJa21