xoseperez/espurna #2453: RF to MQTT

webgps, 2021-06-12:
Does espruna allow to send specific mqtt message on switch on/off via RF? If not would it be possible to do that?

ColinShorts, 2021-06-12:
I use the rfb direct build (requires cutting traces and soldering resistors) with a bunch of door/window sensors, and pir's, and handle the received codes using OpenHAB rules*. I'd imagine that most home automation software will allow you to do the same kind of thing.

*Happy to provide examples of you're using OpenHAB, but I'm still on 2.5 until I find the time to start afresh 🙂

davebuk, 2021-07-19:
See #2296, it might help.

Like @ColinShorts I use the RF code sent by espurna via mqtt in openHAB. As an example I use the following rule and configure the last 6 digits of the message to trigger an event.
```
rule "DevRF001 MQTT"
	when
		Item Dev001rf changed
	then
		var String Devrf_string = Dev001rf.state.toString
		Devrf_string=Devrf_string.substring(10, Dev001rf.state.toString.length)
		logInfo("Log", Devrf_string)
		if (Devrf_string == "0F9190") 
			{sendBroadcastNotification("TEST OPEN")
			}
		if (Devrf_string == "07F80E") 
			{sendBroadcastNotification("TEST CLOSED")
			}
end
```

ColinShorts, 2021-07-20:
I did completely forget about this, but I suspect @webgps has too?

I've got a couple of different magnetic switches, and an rf remote dummy switch in my upstairs office... I've cut down the examples a chunk, but otherwise I've pretty much used the same logic as @davebuk, using switch/case instead of multiple if's. I've used asterisk to obscure some of the codes, just because. ;)

```
rule "RFBdirect received update"

when
    Item rfbdirect_rfin received update
then
    val inStr = rfbdirect_rfin.state.toString
    val lastSixChars = rfbdirect_rfin.state.toString.substring(12,18)
    switch (lastSixChars) {
        case "***59E" : {
            //Ensuite window opened
            postUpdate(Switch_SF_EnsuiteWindow,OFF)
            if (Fan_SF_Ensuite.state == ON) {
                sendCommand(Fan_SF_Ensuite,OFF)
            }
        }
        case "***597" : {
            //Ensuite window closed
            postUpdate(Switch_SF_EnsuiteWindow,ON)
        }
        case "***59B" : {
            //Ensuite window tamper
            postUpdate(Switch_SF_EnsuiteWindowTamper,ON)
        }
        case "***21E" : {
            //Office window opened
            postUpdate(Switch_SF_OfficeWindow,OFF)
        }
        case "***217" : {
            //Office window closed
            postUpdate(Switch_SF_OfficeWindow,ON)
        }
        case "***21B" : {
            //Office window tamper
            postUpdate(Switch_SF_OfficeWindowTamper,ON)
        }
        case "***59E" : {
            //kitchen door opened
            Switch_FF_KitchenDoor.postUpdate(OFF)
            say("veranda door opened", "googletts:enGBStandardA", "chromecast:chromecast:5fc93******")
        }
        case "***C08" : {
            //office display
            if (Light_SF_OfficeDisplay_dimmer.state > 0) {
              sendCommand(Light_SF_OfficeDisplay_dimmer,0)
            } else {
              sendCommand(Light_SF_OfficeDisplay_dimmer,ON)
            }
        }
        default : {
            // reported device isn't configured yet.
            logInfo("rfbdirect found an unknown device: ",lastSixChars,"(",inStr,")")
            if (rfbridge_notify.state == ON) {
              sendNotification("my_email@addy.tld", "rfbdirect found an unknown device: " + lastSixChars + "(" + inStr + ")")
            }
        }
    }
end
```

mcspr, 2021-07-21:
Besides using codes directly, existing relay <-> rfbcode mapping already does this

> specific mqtt message on switch on/off via RF

is just '0' (off) or '1' (on) status message for the Nth relay, by default `${root-mqtt-topic}/relay/${n-th-relay}`.

---

Using `relayPayloadOn` and `relayPayloadOff` settings strings, default '0' and '1' can be replaced with a something custom
https://github.com/xoseperez/espurna/blob/8e5ab5c902a23dfd774dd9e768963856d4f26bd3/code/espurna/relay.cpp#L1167-L1169
https://github.com/xoseperez/espurna/blob/8e5ab5c902a23dfd774dd9e768963856d4f26bd3/code/espurna/relay.cpp#L1383-L1386
https://github.com/xoseperez/espurna/blob/8e5ab5c902a23dfd774dd9e768963856d4f26bd3/code/espurna/relay.cpp#L1582
(global for each relay, but I guess another enhancement would be to have per-relay strings)

closed 2021-07-21 by mcspr