xoseperez/espurna #2454: multiple temp sensor correction one by one

djsomi, 2021-06-15:
Hi All,

First of all thank you very much all of you working on this great stuff!

I am using v1.14.1 for multiple temperature sensor readings in one device and pushing them into influxDB.
I faced that my DS18B20 sensors are not accurate. For example I have 3 sensor on one board, measuring the same thing, gives me different values. eg. 21c 23c 20c 

I want to correct them, but if I set correction, it will affect all of them. 

Is there any way to set correction sensor by sensor?

Many Thanks
Zoli

mcspr, 2021-06-16:
> Is there any way to set correction sensor by sensor?

This was added as part of the future 1.15 and is available in nightly builds or by manually building from dev branch. Internally, we use `tmpCorrection` setting as an offset value for every sensor, but now there are also `tmpCorrection0`, `tmpCorrection1`, and etc. per each temperature sensor available. This can be set either with [terminal](https://github.com/xoseperez/espurna/wiki/Terminal) in webui / telnet / serial or through backup .json upload

You should see the index after calling `magnitudes` command in the terminal, or refer to the suffix used in the MQTT (`/temperature/0`, `/temperature/1`, ...)

djsomi, 2021-06-17:
Thanks for the reply, I've tried to build DEV, as I built the release version, and getting the following error:

https://pastebin.com/txj48SVx

Can you help me, what to check?

Thanks
Zoli

mcspr, 2021-06-17:
> warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=0 to SConscript calls.
Missing SConscript 'C:\Users\somog\.platformio\packages\framework-arduinoespressif8266@3.20704.0\tools\platformio-build.py'

Is the file really missing?

PIO issue tracker sometimes suggests to look at antivirus software blocking it's package installations... but I wonder if the downloaded package broken somehow b/c I just noticed invalid ArduinoJson install locally with missing files as well
Also https://github.com/xoseperez/espurna/issues/2455

djsomi, 2021-06-20:
> Is the file really missing?
>
> PIO issue tracker sometimes suggests to look at antivirus software blocking it's package installations... but I wonder if the downloaded package broken somehow b/c I [...]

Solved it: I had to clone the code with git, for some reason if I opened code folder from the "download zip" dropped several random error msg, on a fresh installation also.

So now I can test how the correction values are working.

djsomi, 2021-06-21:
Thanks working fine!

Just a last one: how can I issue tmpCorrection command from webui? I was able to set with backup / restore json.

Thanks!

mcspr, 2021-06-21:
> Thanks working fine!
> 
> Just a last one: how can I issue tmpCorrection command from webui? I was able to set with backup / restore json.
> 
> Thanks!

`set <key> <value>` in the DEBUG panel
`get <key>` to see the value
`del <key>` to remove

closed 2021-06-21 by mcspr