xoseperez/espurna #2455: Compiling error

javirnof, 2021-06-17:
I can't compiling recently dev versions. I've cloned with git. I've tried some commits back but same error. Only last estable version (1.14.1) I can compile without error...

Platformio: PlatformIO Core, version 5.1.1
Visual Studio Code Windows

```
Compiling .pio\build\nodemcu-lolin\src\api_common.cpp.o
In file included from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp:11:0,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/JsonSerializer.hpp:7,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/JsonPrintable.hpp:12,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariantBase.hpp:11,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:13,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../JsonBuffer.hpp:12,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/JsonParser.hpp:7,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/JsonBufferBase.hpp:7,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/DynamicJsonBuffer.hpp:7,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson.hpp:9,
                 from .pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson.h:9,
                 from espurna\settings.h:19,
                 from espurna\gpio.h:11,
                 from espurna\espurna.h:31,
                 from espurna\compat.h:9,
                 from espurna\DebounceEvent.cpp:33:
.pio\libdeps\nodemcu-lolin\ArduinoJson\src/ArduinoJson/Deserialization/../Serialization/../Serialization/FloatParts.hpp:9:41: fatal error: ../TypeTraits/FloatTraits.hpp: No such file or directory
```

mcspr, 2021-06-17:
What is the content of .pio\libdeps\nodemcu-lolin\ArduinoJson\.piopm and .pio\libdeps\nodemcu-lolin\ArduinoJson\library.json?

PIO also suggests changing our spec to a namespaced one, wonder if it's just a wrong package from some mirror that is putting it instead the original one. Does this work?
```diff
diff --git a/code/platformio.ini b/code/platformio.ini
index b76ade06..53e04ac1 100644
--- a/code/platformio.ini
+++ b/code/platformio.ini
@@ -128,7 +128,7 @@ git_platform_packages =

 lib_deps =
     https://github.com/me-no-dev/ESPAsyncTCP.git#7e9ed22
-    ArduinoJson@5.13.4
+    bblanchon/ArduinoJson@5.13.4
     https://github.com/marvinroger/async-mqtt-client#v0.8.1
     https://github.com/xoseperez/eeprom_rotate#0.9.2
     https://github.com/me-no-dev/ESPAsyncWebServer#b0c6144
```

javirnof, 2021-06-18:
Hi. After some investigating platformio wasn't downloading deps... I've reinstalled platformio, clean clone... and magic... it'works! Thanks

closed 2021-06-18 by javirnof