xoseperez/espurna #2457: MQTT heartbeat status issue.

davebuk, 2021-06-26:
Two small issues with the latest build.
The `status` MQTT is sending the root topic twice.
`[MQTT] Sending WiFi004/WiFi004/status => 1`
The other status messages are fine.
`[MQTT] Sending WiFi004/version => 1.15.0-dev.git44298d40`

Also, the core version isn't showing in the webUI or the terminal when I send `info`.

mcspr, 2021-06-27:
> The status MQTT is sending the root topic twice.
[MQTT] Sending WiFi004/WiFi004/status => 1

see the commit

> Also, the core version isn't showing in the webUI or the terminal when I send info.

when building what core version is used? release 3.0.0 has bugged version string, but should be fixed with 3.0.1

closed 2021-06-27 by mcspr

davebuk, 2021-06-27:
Building with `latest` so 3.0.0.

Commit fixed the issue. Thanks.

commit e6b6a691 (2021-06-27): mqtt: do not compose will topic twice

fix #2457