xoseperez/espurna #2459: Schedule UI is broken

Sag06, 2021-07-03:
06/30/2021 dev compiled here.

Just one switch schedule seems to work.  But if I try to set another switch schedule, “Use UTC time”, “Restore on boot” and “Enabled” buttons from second schedule toggle buttons on first schedule.  Same mess if more than two schedule.

Deleting shedules seem to be broken too. No schedule are shown on UI after deleting them, but if I backup json I can see shedules remain.

mcspr, 2021-07-04:
> Just one switch schedule seems to work. But if I try to set another switch schedule, “Use UTC time”, “Restore on boot” and “Enabled” buttons from second schedule toggle buttons on first schedule. Same mess if more than two schedule.

see the commit

> Deleting shedules seem to be broken too. No schedule are shown on UI after deleting them, but if I backup json I can see shedules remain.

even after rebooting?

closed 2021-07-04 by mcspr

mcspr, 2021-07-05:
... and 50f247bd185d1067add1d6a2d087b6a296242b91 also addresses deleting everything. It still needs a reboot to do the full clean-up though

Sag06, 2021-07-07:
Half solved.  07/06/2021 comp.
Now I can add scheduled tasks without messing buttons and save them. They are on json backup and they work. Good.
But on Espurna Web UI I can't see them, even after rebooting. Only green "add switch schedule". Sorry.

mcspr, 2021-07-07:
does adding comma here show them? there is console error in the browser's js console
```diff
diff --git a/code/espurna/scheduler.cpp b/code/espurna/scheduler.cpp
index 700d71b3..d29a99de 100644
--- a/code/espurna/scheduler.cpp
+++ b/code/espurna/scheduler.cpp
@@ -196,7 +196,7 @@ const char* const schema_keys[] PROGMEM = {
     "schAction",
     "schRestore",
     "schUTC",
-    "schWDs"
+    "schWDs",
     "schHour",
     "schMinute",
 };
```

Sag06, 2021-07-08:
That did the trick. But "Enabled" buttons appear "No" in spite "schEnabled" keys are "1" on json backup. Changing UI  Enabled buttons to "Yes" led to "No changes detected".

mcspr, 2021-07-08:
> That did the trick. But "Enabled" buttons appear "No" in spite "schEnabled" keys are "1" on json backup. Changing UI Enabled buttons to "Yes" led to "No changes detected".

Hm. Missed in the .js change (and not really tested properly, as it turns out). Changing .js code is an adventure every time, as the same code for other inputs does not work with toggle buttons :)

Sag06, 2021-07-16:
Any chance to fix that last schedule issue?

mcspr, 2021-07-16:
WIP
Checkboxes are also broken in case you delete something in the middle and add a new schedule. The underlying problem is \<label for=...\> tags and respective \<input name=...\> need to be re-generated every time schedules list changes, but I actually ended up rewriting more than just solving that issue alone so I need some time to test it

commit 526a5dd2 (2021-07-04): webui: always initialize checkbox with the unique id

even when the cfg object does not list it, like with scheduler button
trying to fix #2459