xoseperez/espurna #2460: 9 switches Max are exposed to Alexa

Sag06, 2021-07-16:
Sonoff Bridge Direct Hack. Dev comp 07/06/2021. FauxmoEsp updated 3.4.0.
A.- 19 switches (DUMMY_RELAY_COUNT   19):  19 switches discovered by hass. NONE by Alexa.
B.- 9 switches. (DUMMY_RELAY_COUNT   9):  9 switches discovered by hass. 9 discovered by Alexa.
Two digits problem? Same issue was present some months ago with hass discovery.

Sag06, 2021-07-16:
Also, on ESPURNA Web, switches (switch configuration), switches are no listed. IE instead of  Switch #0 (2), Switch #1 (2), Switch #2 (2), I can only see (), (), () on a 19 switches compilation.

mcspr, 2021-07-16:
related to https://github.com/vintlabs/fauxmoESP/issues/152, but no interest on that front so far from the users or developer of the lib