xoseperez/espurna #2462: Using a domoticz device as sensor.

HilverinkJ, 2021-07-30:
Hi,
I have the following question;
Can I use a domoticz (sensor) device and switch a relay when reaching a threshold?

Joop.

mcspr, 2021-08-01:
Can you explain a bit more? Domoticz (sensor) device, meaning something only existing on the Domoticz server, or the ESPurna sensor entity and could trigger some action upon reaching the specified threshold?
Doesn't the https://github.com/xoseperez/espurna/wiki/RPN-Rules cover this use-case though `$sensor_magnitude_name` variables?

HilverinkJ, 2021-08-02:
At te moment I'm switching the circulation pump of my floorheating system On/Off with a DS18B20 sensor depanding on RPN-Rules.
[image]
[image]

Now I like to do the same with my zone heatingsystem. But I like to use a virtual (room temp) sensor existing on the Domoticz server to switch On/Off a corresponding zone unit.
[image]

mcspr, 2021-08-03:
With existing switches code we listen to the `domoticz/out`, but there does not seem to be any sensor updates published from the dcz side [(or any mention in the wiki about this use-case, where we want domoticz to publish data and not receive some input)](https://www.domoticz.com/wiki/MQTT)

Quick test with:
- create mqtt client device
- create dummy switch
- create virtual temp sensor
- update through `curl '127.0.0.1:8080/json.htm?type=command&param=udevice&idx=2&nvalue=0&svalue=31'`

Adding 'notification' for the device also does not do anything but notify the browser window

closed 2022-05-05 by mcspr