xoseperez/espurna #2463: Magichome controller using wrong channels for RGB

davidcoulson, 2021-07-30:
I flashed my controller and it is showing as having 4 channels, although it only supports RGB (no W channel). Channel 1,2,3 appear to correctly map to R, G, B respectively as identified in hardware.h. However, if I use the RGB picker the device is using channels 0,1,2 for R, G B, vs 1,2,3

How do I go about re-mapping my RGB from 012 to 123?

[025711] [MQTT] Sending LEDTestStrip/channel/0 => 255 (PID 1)
[025712] [MQTT] Sending LEDTestStrip/channel/1 => 255 (PID 1)
[025713] [MQTT] Sending LEDTestStrip/channel/2 => 0 (PID 1)
[025717] [MQTT] Sending LEDTestStrip/channel/3 => 0 (PID 1)
[025722] [MQTT] Sending LEDTestStrip/brightness => 255 (PID 1)
[025728] [MQTT] Sending LEDTestStrip/light_json => {"state":"ON","brightness":255,"color":{"r":255,"g":255,"b":0}} (PID 1)

mcspr, 2021-07-30:
From the hardware.h config, order of channels always should map 0 -> R, 1 -> G and 2 -> B. Removing the extra channel and shifting numbers down should work?
Notice the `LIGHT_CH1_PIN` is channel 0, `LIGHT_CH2_PIN` is channel 1, and etc. it's not the ID number, but the Nth channel

davidcoulson, 2021-08-02:
Awesome - This worked great. Thanks!

closed 2021-08-02 by davidcoulson