xoseperez/espurna #2464: ota is not working

Sag06, 2021-08-19:
08/19/2021 dev comp. Light Core build for two steps upgrading. It gets wifi IP but ota command does nothing. I Bricked one Sonoff S-26 and two  Sonoff Mini that were running Espurna. I had to serial flash to bring them to life again. 
I can add that 07/24/2021 light core works like a charm, but something must be broken after that.

mcspr, 2021-08-19:
Does this fix the issue?
```diff
diff --git a/code/espurna/ota_asynctcp.cpp b/code/espurna/ota_asynctcp.cpp
index d7588168..ebf96124 100644
--- a/code/espurna/ota_asynctcp.cpp
+++ b/code/espurna/ota_asynctcp.cpp
@@ -67,7 +67,7 @@ void writeHeaders(BasicHttpClient& client) {

     headers += F("GET ");
     headers += client.url.path;
-    headers += F( "HTTP/1.1");
+    headers += F(" HTTP/1.1");
     headers += F("\r\n");

     headers += F("Host: ");
```

Sag06, 2021-08-20:
I'll be out home until 24th. Once back I'll flash new devices and post results.

closed 2021-08-20 by mcspr

Sag06, 2021-08-24:
That was the trick.
Ota works again.
Good job mate!