xoseperez/espurna #2465: Initial user password change page doesn't work

Sag06, 2021-08-19:
I upgraded some Sonoff devices to new 08/18/21 dev build and no problem with previous stored user password.

But I flashed two new devices from Ewelink to ESPURNA and to my surprise I couldn't  change default password. Page presented to change password has an array problem on actual dev.  I tried to change password from  PC and from Android Phone, just to be sure it fails. I tried some different passwords too.

I had to flash an old 05/01/2021 comp that allowed me to change defalult password. After that I did a two steps upgrading.

mcspr, 2021-08-19:
> Page presented to change password has an array problem on actual dev. I tried to change password from PC and from Android Phone, just to be sure it fails. 

Does it visually display some kind of message with the reason it fails? Or it simply does nothing?

Sag06, 2021-08-20:
Yes, some visual error message on top, but I forgot to save it. Silly me!  Something about array data.
I tried a 07/24/2021 comp and was faulty too. Then a  05/01/2021 comp which worked fine.
I'll be out home until 24th. Once back I'll flash and post message.

mcspr, 2021-08-21:
> Yes, some visual error message on top, but I forgot to save it. Silly me! Something about array data.
> I tried a 07/24/2021 comp and was faulty too. Then a 05/01/2021 comp which worked fine.
> I'll be out home until 24th. Once back I'll flash and post message.

screenshot of the error is also ok, plus browser info since I can't reproduce the issue with Chrome

errolt, 2021-08-23:
I have pinpointed the issue. The regex for checking the password validity is wrong. There is a "t" at the end of the regex. All passwords must now end with a "t" to be a valid password.

Sag06, 2021-08-24:
@errolt is right. No new password is valid without a "t" at the end.
But even with a valid password (*******t), this is error message I get:

TypeError: Cannot read property 'insertRule' of null
    at moduleVisible (http://192.168.4.1/:17:24393)
    at http://192.168.4.1/:17:37217
    at Array.forEach (<anonymous>)
    at processData (http://192.168.4.1/:17:34619)
    at WebSocket.onWebSocketMessage (http://192.168.4.1/:17:38087)

For more info see the Developer Tools console.

mcspr, 2021-08-24:
What is the browser name & os & version with both errors?
Was there a different error before that though? moduleVisible() change with insertRule has happened after the report -> https://github.com/xoseperez/espurna/commit/03573b783e10c36d9ff0283ec07d3cb9ee726f1b, on 22/08 and not the previous 18/08 build from the original message
(...and that's whole other thing to understand wth is with the css updates...)

Sag06, 2021-08-24:
1. Erased flash from serial.
2. Flashed  compiled Espurna.
3. Connected to 192.168.4.1.
4. Procced to change password.
5. Message new password is not valid.
6. Added "t" to the end of new password.
7. Password saved.
8. User and password login
9. After login, error message I posted.
10.  Chrome browser. From Android and from Windows 10.

Sag06, 2021-08-24:
[image]
[image]
[image]
[image]

Sag06, 2021-08-24:
Forgot to mention I compiled again actual dev code

Sag06, 2021-08-24:
I tried 24/07.  Password end must be "t", but no error after saving it.
On 01/05 password didn't need final "t".
Hope it helps a little.

commit 31688f1b (2021-08-20): ota: missing space in the outgoing http request

fix #2465