xoseperez/espurna #2466: Master / slave option for syncing without mqtt?

CrispinP, 2021-08-20:
I'm moving out of my house soon and have many coloured leds which each use their own controller. The kitchen, for example, has 7 controllers.

Normally the above was managed by openhab via mqtt but I'm not leaving that behind as the new owner is not tech savvy.

What I'd like to do is leave the controllers synced so he can only connect to one via his phone and the rest follow. He'd obviously have to have them all connected to his wifi.

I was thinking of using the api to poll for switch and colour settings from host / master. Is anything like this possible or in the pipeline. If not, I'll take a look at it.

C

mcspr, 2021-08-20:
Something like this, based on multicast?
https://github.com/xoseperez/espurna/pull/2355

Other option is to remove openhab-like software from the stack, but keep MQTT server on a small'ish device (openwrt, something custom) that is running the WiFi AP? I have also reworked the relays group thing to have separate pub and sub topics, which I think would make sense for lights as well? Likely, with a different payload format too.
(plus, parsing / handling received topics with MQTT wildcards are also supported from the sw side)

CrispinP, 2021-08-20:
I suspect the multicast might be a better option as it can be broadcast and anyone listening can act. I think that is how it works? Sorry, not very familiar.

I've thought of leaving an old RPi running with mosquitto on it as a solution. I guess that'll be ok for a year or so before the SD card messes up. 

I'm in two minds - someone who manages to install espurina on a device to control their LEDs should be able to also setup HASSIO, OpenHAB etc. On the otherhand though, a way to make it simple via multicast also has benefits.

mcspr, 2021-08-21:
> edit: How does #2355 work once implemented?

Standalone e.31 software (PC, phone, etc.) will send packet containing certain channel values. From our side we associate e.31 id with the channel id. And our side is receive-only.

> I've thought of leaving an old RPi running with mosquitto on it as a solution. I guess that'll be ok for a year or so before the SD
> card messes up.
>
> I'm in two minds - someone who manages to [...]

Or, there is a dedicated master device publishing to what others listen to. Additional topic for the master can also be used for the external control.

Also note that rpi 4 should be able to boot from usb, old ones are a bust though
What I originally though [is something like this](https://openwrt.org/toh/xiaomi/xiaomi_redmi_router_ac2100) with some pre-requisite 'unlocking' of the original software.