xoseperez/espurna #2467: ARILUX_AL_LC06 in schedule loop when setting brightness with WW channel enabled

Dujith, 2021-08-24:
**Bug description:**
When "Use white channel" is enabled any brightness setting below 255 will cause a loop with this message:
`[LIGHT] Scheduled transition for 10 (ms) every 10 (ms)`
The dimmer will stop responding and might restart. Disabling "Use white channel" stops this and i can set brightness.

**Steps to reproduce**
Enter the web interface, goto menu "Lights" and select "YES" at "Use white channel"
Save and reload.
Set either the brightness from the "STATUS" page or via MQTT after that the message
'[LIGHT] Scheduled transition for 10 (ms) every 10 (ms)'
starts looping. Setting a color will stop this if the loop has not run for long.

**Expected behavior**
Setting brightness with "Use white channel" enabled should just set the brightness.

**Device information**
ESPURNA 1.15.0-dev built 2021-08-23 20:39:05
mcu: esp8266 chipid: 8CCE4E03F316
sdk: 2.2.2-dev(38a443e) core: 3.0.2
md5: 06c2763e137e567db3d138dee7f2cebf
support: API BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS MQTT NTP ARDUINO_OTA OTA_CLIENT RELAY SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
last reset reason: External System
extra info: External System
+OK

**Crash report**
No crash report.

**Tools used**
* *Windows 10 Pro 10.0.19043 Build 19043*
* *Chrome 92.0.4515.159*
* *Arduino 1.8.15*

mcspr, 2021-08-24:
> When "Use white channel" is enabled any brightness setting below 255 will cause a loop with this message:
[LIGHT] Scheduled transition for 10 (ms) every 10 (ms)
The dimmer will stop responding and might restart. Disabling "Use white channel" stops this and i can set brightness.

yep, that's a bug from here on:
https://github.com/xoseperez/espurna/blob/fd913f15a1e1aa5bf6a2bbe9e0d6e9207f6f0db6/code/espurna/light.cpp#L755-L780

straightforward fix is to replace `.value` with `.inputValue` in these conversions, which will no longer make this dependent on the previous transition / brightness application. but, not yet sure if the math actually checks out.

mcspr, 2021-08-31:
...also the latest commit(s) slightly increases the values of all the channels, as the original calculations seemed to mess up the integral vs. floating-point math of the factor (at line 763 above) . But, I have no rig to test this atm, only looking at the new numbers