xoseperez/espurna #2468: espurna\debug.cpp:402:6: warning

davebuk, 2021-08-29:
Hi

I have just been running some builds with 1.15.0-dev.gita55cf0b and sometimes get the warning below.

The build completes and the firmware seems to run ok, just wandered what the warning was about? `espurna\debug.cpp` around line 400 references `// namespace buffer`. 

```
espurna\debug.cpp:402:6: warning: 'void debug::{anonymous}::output(Print&, const char (&)[10], const char*, size_t)' defined but not used [-Wunused-function]
  402 | void output(Print& out, const char (&prefix)[10], const char* message, size_t len) {
      |      ^~~~~~
```

mcspr, 2021-08-30:
> The build completes and the firmware seems to run ok, just wandered what the warning was about?

The only place in debug.cpp that actually uses the function may be hidden by the `DEBUG_SERIAL_SUPPORT` flag set to zero:
https://github.com/xoseperez/espurna/blob/a55cf0b8470c33cf26e890f1167adbd46f168e30/code/espurna/debug.cpp#L485-L487
b/c of the recent changes adding `{anonymous}` namespace, compiler gives a cue that there may be some logical errors in the file *(...and even more technical stuff related to how the binary files are constructed...)*. In this case it's not though, since it is intended to disable the code. But, the function should be hidden too, if that's the case.

closed 2021-08-31 by mcspr

davebuk, 2021-09-01:
It could have been one of the builds where `DEBUG_SERIAL_SUPPORT` flag was set to zero.

I've just tried one of those builds and others and I haven't had the warning.

commit 1ae7f272 (2021-08-31): debug: fix build warning with serial disabled

resolve #2468