xoseperez/espurna #2470: Webui. Switches page doesn't work

Sag06, 2021-09-08:
Switches page (Switch/relay configuration) always says "No changes detected" after saving changed values.  Boot mode or Pulse mode values i.e.

Tried on Sonoff Mini and Blitzwolf BW-SHP6.
Actual dev code.
Updated Chrome Browser. From PC and Android mobile.

mcspr, 2021-09-09:
Since https://github.com/xoseperez/espurna/commit/fa3deeffbfa622ecd1869af2563940fb3143e94e for the `change` event handler to appear, WebUI needs input or select to already exist in .html or the container element must have `settings-group`:
https://github.com/xoseperez/espurna/blob/f748b6487ca172824266c873916e76786a0080df/code/html/custom.js#L2649-L2650
https://github.com/xoseperez/espurna/blob/f748b6487ca172824266c873916e76786a0080df/code/html/index.html#L920
Where relay config does not have it, so no input is processed from there
https://github.com/xoseperez/espurna/blob/f748b6487ca172824266c873916e76786a0080df/code/html/index.html#L545

Sag06, 2021-09-19:
I'm sorry, but I'm not sure to understand you.  Switches page on Web interface is useless now because we can't change values. No changes are detected in spite switch/gpio are correct on this page.

mcspr, 2021-09-19:
Just a note for myself or anyone that may've seen the issue.

> Switches page on Web interface is useless now because we can't change values. No changes are detected in spite switch/gpio are correct on this page.

It is a bug, no doubt. Just haven't got around pushing the update.

Sag06, 2021-09-19:
Don't worry. We can wait.
Thank you very much.

commit 444fcec8 (2021-10-03): webui: fix relay and curtain config change detection

resolve #2470
closed 2021-10-03 by mcspr