xoseperez/espurna #2472: can't use gpio 16 for LED channels (RGBWW)

hamed-ta, 2021-09-24:
Hello,

I have custom hardware for an RGBWW LED light which I used gpio 16 for red LEDs but, when I compile the code with these configs

    #define LIGHT_CH1_PIN       16      // RED
    #define LIGHT_CH2_PIN       14      // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
    #define LIGHT_CH4_PIN       15      // COLD WHITE
    #define LIGHT_CH5_PIN       13      // WARM WHITE

esp keep restarting after boot but if I change the channel 1 pin for example to gpio 4, it works without the red channel.

what is the problem?

thanks

mcspr, 2021-09-24:
register definitions end at GPIO15, it's out of bounds access error that was not checked by the code
https://github.com/xoseperez/espurna/blob/9087fcf618538830dbf73e5cc6ea7c82db8a09c9/code/espurna/light.cpp#L2908-L2921

pwm on gpio16 is not supported by the built-in implementation though, gpio16 requires special access register and some code updates so the masking is not limited to the first 16bits
something like the current Core implementation instead what we have right now
https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/cores/esp8266/esp8266_peri.h#L47-L57
https://github.com/esp8266/Arduino/blob/60fe7b4ca8cdca25366af8a7c0a7b70d32c797f8/cores/esp8266/esp8266_peri.h#L125-L136
https://github.com/xoseperez/espurna/blob/9087fcf618538830dbf73e5cc6ea7c82db8a09c9/code/espurna/pwm.c#L88
https://github.com/esp8266/Arduino/blob/25e1b3b61c6e37a648e49292916f4ed672304dfb/cores/esp8266/core_esp8266_waveform_pwm.cpp#L542
(or, just use analogWrite so it's used instead of the pwm.c, that's also an option)

hamed-ta, 2021-09-24:
thanks for the explanation,

are you going to fix it?

mcspr, 2021-09-24:
for the specific board you can modify light.cpp, the gist of it is:
- remove pwm_... init and start, leave just the pinMode(pin, OUTPUT)
- replace pwm_set_duty bits in the LIGHT\_PROVIDER\_DIMMER portions with analogWrite

if that works ok, let's add it here as a 2nd pwm option

hamed-ta, 2021-09-24:
Ok, let me test it.

hamed-ta, 2021-09-24:
so it's now this way, is that correct?

[image]

[image]

mcspr, 2021-09-24:
\* `analogWrite(_light_channels[channel].pin, pwm)`

and there's a 2nd pwm_start in _lightProviderHandleUpdate

mcspr, 2021-09-24:
ah, and notice that value upper range is set via `LIGHT_MAX_PWM`, right now it is the default 10000 which should be 255 to fit the Core value range (or it would always be at maximum brightness)

hamed-ta, 2021-09-24:
now it work well,

hamed-ta, 2021-09-24:
> ah, and notice that value upper range is set via `LIGHT_MAX_PWM`, right now it is the default 10000 which should be 255 to fit the Core value range (or it would always be at maximum brightness)

I didn't change it, do I need to do it while it's working?

mcspr, 2021-09-24:
> > ah, and notice that value upper range is set via `LIGHT_MAX_PWM`, right now it is the default 10000 which should be 255 to fit the Core value range (or it would always be at maximum brightness)
> 
> I didn't change it, do I need to do it while it's working?

would be a more precise control, right now from our side it converts from 0...255 (channel value) to 0...10000 (pwm), and analogWrite will cap things at it's internal 'scale' limit, so low is too low and high is too high. what limit is depends on the version:
https://github.com/esp8266/Arduino/blob/2.7.4/cores/esp8266/core_esp8266_wiring_pwm.cpp#L57 - 1023
https://github.com/esp8266/Arduino/blob/3.0.2/cores/esp8266/core_esp8266_wiring_pwm.cpp#L58 - 255

hamed-ta, 2021-09-24:
> would be a more precise control, right now from our side it converts from 0...255 (channel value) to 0...10000 (pwm), and analogWrite will cap things at it's internal 'scale' limit, so low is too [...]

I changed the LIGHT_MAX_PWM to 255 and now the light on max brightness is too low, I think the max should be 1000.

hamed-ta, 2021-09-24:
there is a problem in UI, when I choose a color from the color wheel, it jumps to the opposite side and chooses another color, why is that?

https://user-images.githubusercontent.com/9883945/134660830-0a6bb984-82c9-4fe4-a14d-1962ad59ae49.mp4

hamed-ta, 2021-09-25:
Yes, solved, can I make a pull request for this?

hamed-ta, 2021-09-25:
There is the same problem with _lightFromHsvPayload, I changed the orders of values 1 and 2 but it didn't make any change.

https://github.com/xoseperez/espurna/blob/9087fcf618538830dbf73e5cc6ea7c82db8a09c9/code/espurna/light.cpp#L1092

mcspr, 2021-09-25:
> There is the same problem with _lightFromHsvPayload, I changed the orders of values 1 and 2 but it didn't make any change.
> 
> https://github.com/xoseperez/espurna/blob/9087fcf618538830dbf73e5cc6ea7c82db8a09c9/code/espurna/light.cpp#L1088-L1094

Current HSV calculation is made using the '.target' (i.e. what mapping red(), green() and blue() return) value of the channel, which is set *after* brightness function is applied to the '.input' aka raw value set for the channel. I have not actually looked at how it *may* be done through the channel's '.input' exclusively, because of 'Value' part of the HSV

hamed-ta, 2021-09-27:
> > There is the same problem with _lightFromHsvPayload, I changed the orders of values 1 and 2 but it didn't make any change.
> > https://github.com/xoseperez/espurna/blob/9087fcf618538830dbf73e5cc6ea7c82db8a09c9/code/espurna/light.cpp#L1088-L1094
> 
> Current HSV calculation returned back to the UI is made using the '.target' (i.e. what mapping red(), green() and blue() return) value of the channel, which is set _after_ brightness function is applied to the '.input' aka raw value set for the channel. I have not actually looked at how it _may_ be done through the channel's '.input' exclusively, because of 'Value' part of the HSV

Ok, I will check to see if I can solve the problem.

hamed-ta, 2021-12-21:
Hello, 

> for the specific board you can modify light.cpp, the gist of it is:
> 
> * remove pwm_... init and start, leave just the pinMode(pin, OUTPUT)
> * replace pwm_set_duty bits in the LIGHT_PROVIDER_DIMMER portions with analogWrite
> 
> if that works ok, let's add it here as a 2nd pwm option

Hello, this approach doesn't work anymore! did you make any change in the code?

hamed-ta, 2021-12-21:
if using GPIO16 is problematic, can I use RXD pin for one PWM channel and while using TXD pin to send debug information over serial?

mcspr, 2021-12-21:
> Hello, this approach doesn't work anymore! did you make any changes to the code?

just https://github.com/xoseperez/espurna/commit/b9db53595056efe0d0cd012913245cd066eb799f to check whether the pin number can be used to access io{func,mux} arrays
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/light.cpp#L3216-L3218
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/light.cpp#L2970-L2972
(and `gpioValid(16)` is still returning `true`)

> if using GPIO16 is problematic, can I use RXD pin for one PWM channel and while using TXD pin to send debug information over serial?

`-DDEBUG_PORT=Serial1` to use GPIO2 as output-only is also an option (but note that it needs to be high, since it is also used for boot mode selection in addition to GPIO0)

Arduino's serial options are peculiar though, I think one needs to call begin() with mode=`SERIAL_TX_ONLY`
https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial
https://github.com/esp8266/Arduino/blob/b5f3d1d6d57281642f3e034a215146da61b0771e/cores/esp8266/HardwareSerial.h#L82
i.e. change our `DEBUG_PORT.begin(DEBUG_SERIAL_BAUDRATE, SERIAL_8N1, SERIAL_TX_ONLY)` in the debug.cpp

hamed-ta, 2021-12-21:
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/light.cpp#L2963-L2968

The problem is here, there is no definition of GPIO16!

mcspr, 2021-12-21:
yes, hence the check :)
```cpp
              io_info[index][0] = _light_iomux[pin]; << out-of-bounds
              io_info[index][1] = _light_iofunc[pin]; << out-of-bounds
              io_info[index][2] = pin; // << out-of-bounds to the pwm lib, writing invalid address
```

`analogWrite` does not need the check though, and the setup above is supposed to be commented out

hamed-ta, 2021-12-21:
by the way, analogWrite seems to work smoother!

mcspr, 2021-12-21:
smoother transitions between brightness levels?
afaik analogWrite has frequency is 1khz vs 10khz new_pwm uses, and there's slightly different scaling (as commented above analogWriteResolution 8bit vs. 10bit) when it comes to setting the exact duty cycle values

hamed-ta, 2021-12-21:
> smoother transitions between brightness levels? afaik analogWrite has frequency is 1khz vs 10khz new_pwm uses, and there's slightly different scaling (as commented above analogWriteResolution 8bit vs. 10bit) when it comes to setting the exact duty cycle values

Yes with the new PWM library in transitions I experience flickering problems sometimes but with analog write it's smooth and as I know the resolution of analog write is 10bit and the frequency can be adjusted from 1kh to 10kh ?!

mcspr, 2021-12-24:
> 

`analogWrite()` is controlled through `analogWriteFrequency(1000)` (aka 1khz in hz)

ours may be 2khz not 10khz, like I assumed from the number
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/config/general.h#L1153
changing the flag would init new_pwm with a different frequency

another point of refactoring would be to have something more self-explanatory for these build flags / settings

closed 2022-10-28 by mcspr