xoseperez/espurna #2476: LED Pattern doesn't seem to trigger second sequence.

davebuk, 2021-09-29:
I have just built using: https://github.com/xoseperez/espurna/commit/62d83c6194630076fa198ddbf0a899ae7fb79672 but my LED pattern doesn't get to the second part of the sequence. It used to work on a 1.14 build.

I have loaded: `[386281] > ledPattern1 => "1000,500,7 500,500,10 300,300,0"` and when the LED is triggered by a `FOLLOW ME` dummy relay, I get the 1000,500 pulses but not the 500,500 after 7 inital pulses.

I tried `set ledPattern1 "100,500,2 500,500,10 300,300,0"`, `reload` and the first pulse length changed but still didn't move onto the second sequence after the now 2 pulses.

davebuk, 2021-09-29:
In addition with this build, the LED `MODE` in the webUI doesn't reflect the actual mode of the LED. My LED #1 shows `Wifi Status` but is actually `> ledMode1 => "2"`.

mcspr, 2021-09-29:
Looks like a wrong condition check here:
https://github.com/xoseperez/espurna/blob/62d83c6194630076fa198ddbf0a899ae7fb79672/code/espurna/led.cpp#L722

Which does not handle finite pattern.
```
@@ -719,11 +602,11 @@ void run(Led& led) {

         switch (current.mode()) {
         case LedDelayMode::Finite:
-            if (status && current.repeat()) {
+            if (status && !current.repeat()) {
                 if (pattern.last()) {
                     led.status(false);
-                    return false;
                 }
+                return false;
             }
             break;
         case LedDelayMode::Infinite:
```

davebuk, 2021-09-29:
Yep, that works.

BTW, I have to press enter on the keyboard for the data to get sent. the 'SEND' button on the webUI doesn't do anything.

mcspr, 2021-09-30:
> In addition with this build, the LED MODE in the webUI doesn't reflect the actual mode of the LED. My LED #1 shows Wifi Status but is actually > ledMode1 => "2".

ledMode and ledInv are swapped in the payload. Will update with a more consistent way to set the actual schema on the same line
https://github.com/xoseperez/espurna/blob/62d83c6194630076fa198ddbf0a899ae7fb79672/code/espurna/led.cpp#L948

> BTW, I have to press enter on the keyboard for the data to get sent. the 'SEND' button on the webUI doesn't do anything.

Ugh. Forms in HTML only "submit" when there's a button with type=submit, type=button doesn't work :/
https://github.com/xoseperez/espurna/blob/62d83c6194630076fa198ddbf0a899ae7fb79672/code/html/index.html#L1699
(and I also only used Enter, even on the mobile...)

mcspr, 2021-10-03:
I think all 3 issues are covered? Notice that there's also `led 0 "100,100,0"` command to apply pattern until the next settings 'reload'

closed 2021-10-03 by mcspr

davebuk, 2021-10-04:
All working, thanks.