xoseperez/espurna #2477: Serial Sensor

pbek, 2021-10-02:
I'm using the wonderful Espurna firmware on my cheap 5$ Sonoff devices (230V, 1 LED, 1 relays, 1 button) for some years to control all kind of stuff. The limitation of those devices is that they only have one extra GPIO pin routed on the board.

So I'm plugging very cheap Arduino Nano (nockoff) boards to the Sonoff boards via serial interface to get more GPIO pins and AD converters from the Arduino (still cheaper than to build everything with a "normal" ESP8266 or ESP32 board).

For that usecase I wrote a small `Serial Sensor` addition to Espurna that just uses some commands like `SENSOR0 1234` (sent by the Arduino via serial) to publish that value 1234 via MQTT.

Is there any interest that I share this addition upstream (just one cpp and h file)?

mcspr, 2021-10-02:
> For that usecase I wrote a small Serial Sensor addition to Espurna that just uses some commands like SENSOR0 1234 (sent by the Arduino via serial) to publish that value 1234 via MQTT.
> Is there any interest that I share this addition upstream (just one cpp and h file)?

Sure. There are even certain custom boards in the tree, so I don't see any harm here.

> So I'm plugging very cheap Arduino Nano (nockoff) boards to the Sonoff boards via serial interface to get more GPIO pins and AD converters from the Arduino (still cheaper than to build everything with a "normal" ESP8266 or ESP32 board).

Notice that recent versions have tried to tackle GPIO HAL, to handle things besides the hardware itself. Only one current example is mcp23xxx digital pins that can be used as relays and buttons (ref. mcp.cpp, gpio.{h,cpp}, relay.cpp, BasePin header). It is not completely done though, since not everything is done by using them. And certain analog pin functions are handled on case-by-case basis (like buttons and i2c extenders)

pbek, 2021-10-03:
The lack of "reachable" GPIO/AD pins on the https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-Basic board drove me to the modification. 😅  There only is GPIO14.

[image]

pbek, 2021-10-03:
Code looks like this: https://github.com/xoseperez/espurna/pull/2478

mcspr, 2021-10-03:
> The lack of "reachable" GPIO/AD pins on the https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-Basic board drove me to the modification. 😅 There only is GPIO14.

What I meant is something like this https://www2.mouser.com/_/?Keyword=io+expander, but those are commonly based on either i2c or spi interfaces. But, can't find anything based off of the uart to 're-use' the protocol. With mcp23xxx mentioned above, we simply read and write u16 number that represents a bitset of turned ON pins as ones and OFF as zeroes.

pbek, 2021-10-03:
I couldn't have used those on my Sonoffs. 😁

mcspr, 2021-10-03:
I feel like there's a translation issue here :)

Consider already existing hardware:

> So I'm plugging very cheap Arduino Nano (nockoff) boards to the Sonoff boards via serial interface to get more GPIO pins and AD converters from the Arduino (still cheaper than to build everything with a "normal" ESP8266 or ESP32 board).

But, imagine these work with a custom binary protocol that allows to ON/OFF individual pins, like the I/O expanders I linked.

pbek, 2021-10-03:
knockoff... sorry. 😁 

> But, imagine these work with a custom binary protocol that allows to ON/OFF individual pins, like the I/O expanders I linked.
edit:... and ofc read their ON or OFF as well

Did mean the Arduino Boards? In my case I wrote a little firmware to read from e.g. the AD and write the value to the serial pins.

davebuk, 2021-10-04:
If you aren't using the LED you can de-solder it and use that GPIO. The same with the relay but needs a little more de-soldering.

pbek, 2021-10-07:
I need both the LED and Relay. 😁

closed 2021-10-12 by mcspr