xoseperez/espurna #2479: arduino-cli: builder does not consider .inc files a valid extension

mcspr, 2021-10-05:
As noticed here:
https://gitter.im/tinkerman-cat/espurna?at=615b7f828c019f0d0b1803d8
Building led.cpp after https://github.com/xoseperez/espurna/commit/de2d5f7350f23c33da11ce174238eef277dc0597 is not possible due to the way arduino-cli handles source files

When starting a build, arduino-cli creates a temporary directory. For example:
```bash
$ ls /tmp/arduino-sketch-F5B33AFB88F05FFFF10421F40C44A86B/
build.options.json     core             espurna.ino.elf  includes.cache  libraries                     local.eagle.flash.ld  sketch
compile_commands.json  espurna.ino.bin  espurna.ino.map  ld_h            local.eagle.app.v6.common.ld  preproc
```

All source files are supposed to be *copied* into a subdir 'sketch', and this dir also contains intermediate .d and the resulting .o. However, only certain files are actually copied there:
```bash
$ ls /tmp/arduino-sketch-F5B33AFB88F05FFFF10421F40C44A86B/sketch/led*
led.cpp                   led.cpp.d                 led.cpp.o                 led.h
```

Current build will fail with a nonsense message on the 'Detecting libraries used...' phase, when it tries to assemble the led.cpp.o
```
$ git clone https://github.com/xoseperez/espurna
$ pushd espurna/code/espurna
$ arduino-cli -v compile -b esp8266:esp8266:nodemcu --build-property "build.extra_flags=\"-DNODEMCU_LOLIN\" -DESP8266"
...
/home/runner/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/include -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/lwip2/include -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino-sketch-F5B33AFB88F05FFFF10421F40C44A86B/core -c -w -Werror=return-type -Os -g -free -fipa-pta -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++17 -ffunction-sections -fdata-sections -fno-exceptions -DMMU_IRAM_SIZE=0x8000 -DMMU_ICACHE_SIZE=0x8000 -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10607 -DARDUINO_ESP8266_NODEMCU_ESP12 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU_ESP12\"" -DFLASHMODE_QIO -DNODEMCU_LOLIN -DESP8266 -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/cores/esp8266 -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/variants/nodemcu -I/home/runner/Arduino/libraries/ArduinoJson/src -I/home/runner/Arduino/libraries/EEPROM_Rotate/src -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/EEPROM -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi/src -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Ticker/src -I/home/runner/Arduino/libraries/espasyncwebserver/src -I/home/runner/Arduino/libraries/ESPAsyncTCP_ID305/src -I/home/runner/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Hash/src -I/home/runner/Arduino/libraries/fauxmoESP/src /tmp/arduino-sketch-F5B33AFB88F05FFFF10421F40C44A86B/sketch/led.cpp -o /dev/null
Alternatives for led_pattern.re.cpp.inc: []
ResolveLibrary(led_pattern.re.cpp.inc)
  -> candidates: []
/home/runner/espurna/code/espurna/led.cpp:393:10: fatal error: led_pattern.re.cpp.inc: No such file or directory
  393 | #include "led_pattern.re.cpp.inc"
```

Current list of valid extensions:
https://github.com/arduino/arduino-cli/blob/f03580ca510aa6ccc8ab256d302bd84830fabe67/arduino/globals/globals.go#L32-L59

mcspr, 2021-10-05:
Also see
https://github.com/arduino/arduino-cli/issues/1149

closed 2021-11-29 by mcspr

Sag06, 2021-12-18:
Today dev code. Arduino IDE 1.8.13. .ipp files are not valid though

exit status 1
led_pattern.re.ipp: No such file or directory
web_asyncwebprint.ipp: No such file or directory

I could compile changing file extensions to .h and include references

mcspr, 2021-12-18:
Does it work with 1.8.16?
Does it work in cli? I do remember testing this :>

Sag06, 2021-12-18:
It doesn't work with 1.8.18. 

Never used pure cli. I'll look how to try it

commit a7a9c504 (2021-11-29): pio: generate .cpp.ipp instead of .cpp.inc

fixes #2479