xoseperez/espurna #2485: Cannot access to ESP01S via Chrome in macosx, but can access in an iPad

Currela, 2021-11-16:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [gitter chat](https://gitter.im/tinkerman-cat/espurna) ([also available with any Matrix client!](https://matrix.to/#/#tinkerman-cat_espurna:gitter.im))*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors help you. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Question**
*A clear and concise description of what the problem/doubt is.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version, if the problem is related to the WebUI*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other information about the problem here.*

Currela, 2021-11-16:
Hi there, I very recently began doing my things in home automation and tried ESPURNA (great system!) but after a few firmwares downloading, tried with ESP01S in a one-relay module. I flashed "espurna-1.14.1-generic-esp01s-relay-40-inv.bin" in it and have problems trying to reach it through a chrome in a macbook (also tried with Safari with same results). After flashing the firmware, I had no problems to reach the ESP8266 in the "ESPURNA XXXX" wifi. I setup an admin passwd and after that, I provide my home wifi SSID and pass and reboot the module. When the ESP01S rebooted, it connected to my home wifi with no problems and then I cannot reach it in chrome (providing the ip assigned), only with the iPad. It's so weird that I have tried everything I can think of but without sucess. I have not this problem with wemos d1 mini, only with the ESP01S. Could you please tell me how can I fix this issue? Thanks in advance,

mcspr, 2021-11-17:
Does it work with a nightly .bin?

Currela, 2021-11-17:
Hi Max. Thanks for your message. I used a "released" .bin found in https://github.com/xoseperez/espurna/releases; I've tried to find the "nightly" firmware but cannot find any nightly files in ESPURNA github (sorry, I'm very newbie with espurna). Please, copy a link where I can get the nightly .bin and I'll use it and feed you back.

Currela, 2021-11-17:
Dear Max. I found the latest nightly .bin (espurna-1.15.0-dev.gitc95ea71c+nightly20211024-generic-esp01s-relay-40-inv.bin). I flashed it, connect in the ESPURNAXXX wifi without any problems, change the admin passwd, give the SSID and pass of my home wifi and reboot the ESP01S and after that, it connected to the wifi without any issues. I tried to reach it via Chrome introducing the IP address the router gave to the ESP01S and cannot reach it neither with Chrome, Firefox or Safari. I can reach it without any problems with my iPad or iPhone with no hassle. I'm a bit astonished with this issue. Any sugestions? Thanks in advance.

mcspr, 2021-11-17:
Anything happens in the web inspector's 'Network' tab? You'd need to check 'Preserve' checkmark, and then reload.
Does the telnet stuff work or the device is completely inaccessible? (`ncat $ip 23`) Note that by default telnet connection is rejected when connected to a WiFi network, but there's a setting on the ADMIN panel.

Currela, 2021-11-17:
Hi Max. Nothing happens in Inspector's Network tab:
[image]
I don't use telnet either.
The most weird thing is that I can reach the chip on Chrome in my iphone and iPad w/o issues.
Thanks for your help.

mcspr, 2021-11-17:
> I don't use telnet either.

What I mean is, you sure IP is actually reachable? Does `ping` respond?

Currela, 2021-11-17:
I ping to the ip (in Terminal):
[image]
But seems is not reachable. The ip is the one the router assigned to the ESP01S and the one I use with the iPad and iPhone is reach it.

mcspr, 2021-11-17:
> But seems is not reachable. The ip is the one the router assigned to the ESP01S and the one I use with the iPad and iPhone is reach it.

Then, I'd advice to look at networking / local firewall or some app blocking access / check whether it is actually in the same network :)
I can't advice on that further, though, since you'd have better understanding of your the networking setup

closed 2021-11-17 by mcspr