xoseperez/espurna #2486: Which version should i use with RF bridge direct hack and ALEXA ?

smcgann99, 2021-11-19:
I am running ESPURNA 1.14.1. RF Bridge

I changed setting to 16 dummy relays, but for some reason Alexa is not discovering anything. 
changed back to 8 and deleted all devices in alexa and now it will not discover the orginal 8 relays ?

Latest "nightly" build seems to be 26 days ago.

mcspr, 2021-11-20:
> I changed setting to 16 dummy relays, but for some reason Alexa is not discovering anything.
changed back to 8 and deleted all devices in alexa and now it will not discover the orginal 8 relays ?

this is a duplicate of https://github.com/xoseperez/espurna/issues/2460
note that both 1.14.1 and nighties still affected by this

> Latest "nightly" build seems to be 26 days ago.

...and the last change was 26 days ago :) [while inheriting the name, it's not the exact same way of doing nightly / daily / neutral build literally every day](https://en.wikipedia.org/wiki/Neutral_build). main idea is to check whether release build will work as it uses the same tools, plus there's a nice side effect of having ready to go binaries

closed 2021-11-20 by mcspr

smcgann99, 2021-11-20:
"main idea is to check whether release build will work"
I'm a bit confused 1.14.1 is the latest release build isn't it ?

mcspr, 2021-11-20:
> "main idea is to check whether release build will work"
I'm a bit confused 1.14.1 is the latest release build isn't it ?

it is. development happens though, nightlies make sure release won't break when it happens