xoseperez/espurna #2487: Parallel installation of two 433MHz rf bridges

tita2000, 2021-11-23:
Hello, I'm wondering if I can setup two SONOFF 433 RF bridges with the exact same configuration to put one ground level of my house and the other one upstairs. Has someone tried this before? Will they interfere with each other, so that the signal gets possible weaker or unreadable? I realize that this might not be the right place and no real issue related post, but I found no other solution. Any comments welcome.
Regards Timo

ColinShorts, 2021-11-23:
I have 2 of these (ground and second floor), but I run one using RFBdirect (traces cut, added resistors). I'm only using them for listening, so the only thing to note is that I often see the same device reporting on both bridges - but as the codes presented are different this doesn't cause me any problems. I'm currently processing the codes using switch/case in openhab rules, but I'm in the slow process of migrating to homeassistant and haven't migrated the sensors yet.

The signals won't be any weaker when you've got multiple rfbridges listening, but if they are reporting to the same endpoint you might need to use some kind of debounce to prevent issues in switches.

tita2000, 2021-11-23:
Thank you. My problem would mostly be sending 433 code to switch devices on or off. Whereby these will be two separate codes, one for on and one for off. So e.g. two bridges will be sending the exact same code simultaneous to switch one 433 power plug.

tita2000, 2021-11-23:
I already thought about using straight 433Mhz rubber antennas and removing the winded coils inside, but I only found solutions for the listening circuit. Will they also work for the sending part of the bridge?

ColinShorts, 2021-11-23:
I'm not sending anything to RF devices, so can't say for certain, but that sounds like it could cause problems if the switch toggles with the same code?

closed 2022-05-05 by mcspr