xoseperez/espurna #2491: Sensor magnitude offset option in webUI not showing.

davebuk, 2021-12-26:
I have a device that is using build: `1.15.0-dev.gitd5273513` with a BME280 sensor and I've noticed that I can't see the section in the webUI for adjusting `PX, TEMP, HUMIDITY` offsets. I know they used to be there in other builds.

I'm going to look to build a firmware tomorrow for another dev board with temp sensor, but before I do, is there something that needs to be adjusted in the code? I've had a quick look through the previous commits and couldn't see anything obvious that may have been fixed since Sep 2021. Thanks.

davebuk, 2021-12-26:
Built a firmware now with latest dev on a wemos d1 mini and get the following error at the top of the page:
```
TypeError: Cannot read properties of undefined (reading 'forEach')
    at updateMagnitudes (http://192.168.1.56/:17:38991)
    at http://192.168.1.56/:17:42063
    at Array.forEach (<anonymous>)
    at processData (http://192.168.1.56/:17:39650)
    at WebSocket.onWebSocketMessage (http://192.168.1.56/:17:43445)

For more info see the Developer Tools console.

```
I had a BME280 connected (which currently has an issue and intermittently fails so data pins disconnected) and I've just added a DS18b20 Waterproof Temperature Sensor but that isn't showing up as a sensor at the moment.

mcspr, 2021-12-26:
2nd one seems to be caused by no sensors being available for reading
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/html/custom.js#L1768
i.e. `data.values` *is* `undefined` where it should've been `[]`, so it's something to fix from both sides (ref. 'Developer Tools / Network' for the /ws endpoint, message causing the error would be `{"magnitudes":{}}`)

1st one is correction aka offset simply missing from the web interface, starting from this commit
d9cb35f7814fdb90ca413f7474f0c915a56cbd24 (pretty long time ago)
it is still handled via `{tmp,press,hum}Correction{index}` settings key, and through the build flags. something to add back, since the data is being sent to ui but not being processed
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/sensor.cpp#L1857-L1865

davebuk, 2021-12-27:
Both sensors are now reading so that error has gone.

Like you said, the `keys` are there for the corrections, just not showing on the webUI.

davebuk, 2022-04-18:
Hi Max. Is this a relatively quick fix? I'm going to build some other devices in the next few weeks and wondered if the fix would be implemented? Thanks.

mcspr, 2022-04-19:
Probably is, but there's also a lot of backlog.
Also a matter of adjusting the .js to read every `magnitudes-setting` value in the payload instead of just Ratio
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/html/custom.js#L2527-L2531
https://github.com/xoseperez/espurna/blob/02581c6988f9d6b52a74c82fb2f6b3d5f895627c/code/html/custom.js#L1895-L1903
(or, at least the correction offsets, which could be just a c/p and changing names of the elements to match the keys used by settings)

closed 2022-05-05 by mcspr

davebuk, 2022-05-05:
Thank you. I'll look to update some of my devices over the weekend.

davebuk, 2022-05-07:
Working, thanks.
Build flags changed and updated.
Just have an issue with I2C bus.

commit 62b2edad (2022-05-05): sns: show and edit correction values in webui

fixes #2491