xoseperez/espurna #2492: SonOff POW R2

JM69330, 2021-12-29:
I have flashed the device with espurna-1.14.1-itead-sonoff-pow-r2.bin image. I can change the status of the switch but the values from the sensors are still zero (voltage, current , power,...).

JM69330, 2021-12-29:
[image]

JM69330, 2021-12-29:
Re-openned

mcspr, 2021-12-29:
Are you sure it is r2? Note of the https://github.com/xoseperez/espurna/issues/1983, I do remember board version confusion in regards to rev2 vs. r2

And the actual pw chip can be seen on the opposite side of the board
https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-POW#flashing
https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-POW-R2

JM69330, 2021-12-29:
Thanks for your reply. Yes it is an R2 - see photo

[image]
o

JM69330, 2021-12-29:
The sensor chip is CS77598-SO, should I try the previous ESPurna version ?

mcspr, 2021-12-29:
Not previous, but the current nightly would be a much better test

Only thing that comes to mind from reading the [cse7766 doc](https://dl.itead.cc/S31/CSE7766.pdf) is to update this line
https://github.com/xoseperez/espurna/blob/ec33ee32422574e769f237e4c5d8c3739ca0b427/code/espurna/sensors/CSE7766Sensor.h#L119

```diff
- _serial->begin(CSE7766_BAUDRATE);
+ _serial->begin(CSE7766_BAUDRATE, SWSERIAL_8E1);
```

JM69330, 2021-12-30:
So far I just flash the .bin so I'm now getting the whole files and rebuild the .bin

JM69330, 2022-01-01:
Fixed !
Not a sw problem. I even checked the sensor chip, the output send data to the ESP. Then I look at the sensor CSE7959B datasheet and I saw the AC line (either the live or neutral) goes to the common so I suspected some internal current loop  then I disconnect the USB adaptor from the the device and the right measurements was displayed correctly. **So my recommendation  : Do not live the USB serial adaptor connected while the AC voltage is applied.**

ColinShorts, 2022-01-01:
The wiki tells us not to connect to these devices using serial when connected to mains power: https://github.com/xoseperez/espurna/wiki/Hardware

This is for safety reasons, you're lucky you didn't fry something (or yourself)

JM69330, 2022-01-02:
Thanks Colin,
I didn't see the wiki in detail. However, even if the USB adaptor was disconneted from the PC but connected to the device the measurements were still corrupted. Only when I disconnected the adaptor from the sonoff device the measures were correct.

closed 2022-01-02 by JM69330