xoseperez/espurna #2497: "Pulse mode on" sonoff 4ch work in reverse

tty74it, 2022-01-05:
I have upgraded to the last  1.15.0-dev.gite271eaa0+nightly20211219
and the sonoff4ch on Switch #0 (gpio) set to  Pulse mode  Normally off  it works the other way around

As workaround  i set  " Pulse mode  Normally  On"  and it works (in reverse )

mcspr, 2022-01-12:
Thanks! I could not reproduce this building locally, but I only now got around picking nightlies and those aren't working as intended :/
I would expect current 'dev' to work correctly, CI troubles need to be resolved first though.

mcspr, 2022-01-14:
using [Jan 14 build](https://github.com/mcspr/espurna-nightly-builder/releases/tag/20220114)
```
get relayPulse0 relayTime0
> relayPulse0 => none (default)
> relayTime0 => 0.000 (default)
+OK
set relayPulse0 1
+OK
reload
+OK
relay 0 1
relay0 {Prov=gpio TargetStatus=on CurrentStatus=off Lock=none}
+OK
[019082] [RELAY] #0 pulse OFF started in 0 (ms)
[019082] [RELAY] #0 set to ON
[019094] [RELAY] #0 set to OFF
```
```
set relayTime0 5
+OK
reload
+OK
relay 0 1
relay0 {Prov=gpio TargetStatus=on CurrentStatus=off Lock=none}
+OK
[134391] [RELAY] #0 pulse OFF started in 5000 (ms)
[134391] [RELAY] #0 set to ON
[135395] [RELAY] Relay mask: 0b1
[139396] [RELAY] #0 set to OFF
[140400] [RELAY] Relay mask: 0b0
```

why the previous nightly failed is another question :shrug:
only apparent difference between versions is u8 -> u32 for the mode type variable. or, I happened to introduce some kind of undefined behaviour when parsing the setting :/

closed 2022-01-14 by mcspr