xoseperez/espurna #2498: How to connect sonoff pow to cloud with espurna

RamrajRS, 2022-01-06:
Hello all.

I have been trying out espurna and was able to access its webserver.
I would like to know if the espurna will enable the sonoff pow to connect my cloud.
Thus, an app can remotely access the module.

Thanks in advance.

mcspr, 2022-01-11:
Not directly, but you can always utilize the fact that MQTT brokers support bi-directional connections - ESPurna device connects to the broker server on LAN, while the broker itself establishes connection to an external server located on the internet

/remote/ at the mosquitto man, for example
https://mosquitto.org/man/mosquitto-conf-5.html

closed 2022-01-11 by mcspr

RamrajRS, 2022-01-12:
I have tried to connect with a cloud mqtt broker. I am not able to get the status connected for mqtt. Please help.