xoseperez/espurna #2499: When using Tone library light PWM doesn't work anymore

hamed-ta, 2022-01-08:
Hello,

My custom hardware has a small speaker connected to GPIO 2, but when I use the Tone library to make some sounds with the speaker, the PWM lights don't work anymore, is there any conflicts between the PWM library and the Tone library and how to solve it in order to use both of them at the same time?

Thanks

mcspr, 2022-01-11:
As mentioned in the #2472, Tone also uses the Core PWM library and hijacks our interrupt handler attached to the NMI TIMER1. I'd expect it to work when using the analogWrite() though?
https://github.com/esp8266/Arduino/blob/9fcf14f81fa9be589530e9596b7c5a264dc81ee8/cores/esp8266/Tone.cpp#L46

Or, in case you don't need too complex tones (or don't need the precision provided by the interrupt handler, and ok with it stalling b/c of the software) - there's also LED patterns set up either manually or through the `ledPattern#` setting where `#` is the nth led Id

For example, in case `ledGpio0 => "2"` / `#define LED1_PIN 2`. In terminal, `led 0 "100,100,10 500,500,0` will cause the LED to blink ON OFF 10 times every 100ms, then indefinitely do a 500ms cycle. Pattern string is in groups of 3 - `TIME_OFF,TIME_ON,NUMBER_OF_CYCLES`

hamed-ta, 2022-01-14:
> As mentioned in the #2472, Tone also uses the Core PWM library and hijacks our interrupt handler attached to the NMI TIMER1. I'd expect it to work when using the analogWrite() though? [...]

Yes, it works with the Core PWM library, but I changed my hardware and stoped using GPIO16 in order to use Espurna PWM library but it conflicts with the Tone library, is there any way that I can use both of them?

Thanks

mcspr, 2022-01-16:
afaik newpwm only implements the analogWrite part with a shared freq, the '50% duty cycle, custom frequency note signal (plus limited duration)' needs to be implemented on top of it / as a separate codepath :/. Core does two separate codepaths depending on what's expected on the pin - the tone() or the analogWrite() - and we can't share the single available hw timer, and the sw part isn't really shareable as we need to select the wake-up time from both

pwm init is also happening manually atm, one would need to edit the io_info list to add an additional 6th PWM channel and then manually call pwm_set_duty(DUTY, 5)

mcspr, 2022-01-19:
...there is also sigma delta generator
https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/SigmaDeltaDemo/SigmaDeltaDemo.ino
but, it is only a single pin and resolution is only 8bit
(and it does not *seem* to be driven by sdk)

hamed-ta, 2022-04-29:
> pwm_set_duty(DUTY, 5

I checked it, the minimum frequency is too high to make any tone.

closed 2022-10-28 by mcspr